Lienol/openwrt #11: passwall 访问控制无效。

Zeal599, 2020-02-12:
passwall 访问控制无效。

AndyWii, 2020-02-13:
Chain PSW (1 references)
target     prot opt source               destination
RETURN     all  --  anywhere             anywhere             match-set laniplist dst
REDIRECT   tcp  --  anywhere             localhost            tcp dpt:7054 redir ports 1041
RETURN     all  --  anywhere             anywhere             match-set vpsiplist dst
RETURN     all  --  anywhere             anywhere             match-set whitelist dst
RETURN     tcp  --  anywhere             120.79.16.255        multiport dports 10724
RETURN     tcp  --  anywhere             139.226.186.59       multiport dports 10502
RETURN     tcp  --  anywhere             139.226.186.59       multiport dports 502
RETURN     tcp  --  anywhere             139.226.186.59       multiport dports 503

localhost 被拒绝因此不能使用GFW 请问大神要怎么处理这个问题！谢谢您！

Lienol, 2020-02-19:
更新最新源码再试下

closed 2020-02-23 by Lienol