Lienol/openwrt #6: 2020.02.08.06版本Passwall无法开启负载均衡功能

sxy79, 2020-02-12:
[image]
[image]
单个节点没有问题，想试一下负载均衡，怎么保存&应用都没反应。手动到“启动项"中开启HAProxy也没有作用，一直显示”未运行“。

AndyWii, 2020-02-14:
我也遇到这个问题，但是我打开了却没有运行数据！[](url)
[image]
[image]

AndyWii, 2020-02-14:
[image]

closed 2020-02-18 by Lienol