Lienol/openwrt #20: Lean-Lede分支，编译报错

chenxiccc, 2020-02-13:
Cleaning module download cache (golang/go#27455)
37666
go: finding github.com/shadowsocks/v2ray-plugin/.go_work/build latest
37667
go: finding github.com/shadowsocks/v2ray-plugin/.go_work latest
37668
can't load package: package github.com/shadowsocks/v2ray-plugin/.go_work/build: unknown import path "github.com/shadowsocks/v2ray-plugin/.go_work/build": cannot find module providing package github.com/shadowsocks/v2ray-plugin/.go_work/build

lonee6, 2020-02-13:
不选中新增加的ss-v2ray插件就行了。

chenxiccc, 2020-02-13:
> 不选中新增加的ss-v2ray插件就行了。

同样的.config，在dev-19.07分支没有问题。可以编译通过。

Lienol, 2020-02-13:
刚同步了大雕源码
git fetch --all && git reset --hard origin/dev-lean-lede
再试试

chenxiccc, 2020-02-13:
> 刚同步了大雕源码
> git fetch --all && git reset --hard origin/dev-lean-lede
> 再试试

Thanks, it works.

closed 2020-02-13 by chenxiccc

sqliuchang, 2020-02-13:
诶，我怎么编译lean分支疯狂报错，19.07一点事都没有…我再试试

sqliuchang, 2020-02-13:
`make[6]: *** No rule to make target '/home/sqliuchang/Lienol/dev-lean-lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-cryptodev-linux-1.10/ioctl.o', needed by '/home/sqliuchang/Lienol/dev-lean-lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-cryptodev-linux-1.10/cryptodev.o'.  Stop.                        Makefile:1520: recipe for target '_module_/home/sqliuchang/Lienol/dev-lean-lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-cryptodev-linux-1.10' failed
`
我的报这个错误…