Lienol/openwrt #35: HC5611的passwall问题，还请大神拔冗看一眼

PaulYoung0265, 2020-02-16:
其实有几个小问题：
1、基于你的代码编译版本安装(mtd 刷写）以后会出现LED灯在启动以后自动关闭，而不是正常黄、绿依次闪烁的问题。我看南浦月的patch讨论里面有人说定义应该是blue而不是green，不知道能否解决这个问题？
2、另外，passwall在5611上表现比较弱，同样的GCP节点几乎同时测，用SSR+大概有9-10M，passwall只能5-6M。不知能否优化改善？
3、能否在你基于L大的代码里面加入对HC5611的支持？
如有闲暇，还请看一眼，多谢多谢！

Lienol, 2020-02-24:
没机器了，无法测试
不过貌似有人发过hc5611的dts

Lienol, 2020-02-29:
更新到最新源码试试
https://github.com/Lienol/openwrt/commit/61b5391cbef4ac82cb2835758748f2ba181401bd

PaulYoung0265, 2020-03-01:
感谢感谢，我尽快更新试试看。
不过我已经通过另外一个方法解决了，就是编译GL-iNet V2的版本，查过资料，这个和极路由Go的芯片、内存、ROM大小都是一样的，或许具备通用性。就冒险试了一下，果然成功了，无非是要在WAN里面把端口改为eth0.1。当然，一样的通病就是LED灯不对，等我试过最新的源码再看看。
再次感谢！

PaulYoung0265, 2020-03-07:
昨天晚上得空编译好了以后进行了测试：
1 、LED灯的问题解决了，能正常红绿（blue)闪烁；非常感谢！！！！
2、WiFI的N模式支持不完全，不能达到300M，只能在144M，不管选20M还是40M的频宽，也不管用哪个信道；不过，这个问题不重要，便携嘛，而且网口是百兆的；
3、测速结果有改善，无论v2ray还是trojan节点，FAST/油管都能最高跑到8M以上（当然节点本身要好）。但是同节点同时间从同IP访问，还是略低于SSR+约10%-20%；
无论如何，这个问题基本解决了，再次感谢Lienol大神！

Lienol, 2020-03-08:
[image]
这玩意还是用老cc流畅，用新版openwrt太卡了

closed 2020-03-08 by Lienol

PaulYoung0265, 2020-03-08:
求版本文件，或者编译方法……
还请大神赐教

PaulYoung0265, 2020-03-09:
15.05的源码已经下了，lienol package也加了，是否还要把target也加上{怎么加啊？}？要不然make
config里好像找不到这个……

Lienol, 2020-03-10:
https://github.com/syb999/openwrt-15.05

PaulYoung0265, 2020-03-10:
[image: image.png]
按照这个git地址下载进行编译，出了点问题。
谷歌了一圈没有找到答案，百般无奈之下又来厚着脸皮求教大神了，这个toolchain里面的patch文件，4.8-linaro
<https://github.com/kkkgo/openwrt/tree/master/toolchain/gcc/patches/4.8-linaro>
/100-uclibc-conf.patch
似乎要补丁的文件没有匹配上，导致patch缺失无法编译下去。
请问可以如何处理？

PaulYoung0265, 2020-03-15:
看见我上次提交的问题后面有更新，我也更新了一下代码重新编译， 4.8-linaro
<https://github.com/kkkgo/openwrt/tree/master/toolchain/gcc/patches/4.8-linaro>
/100-uclibc-conf.patch  报错误的部分过去了
重新编译后又遇到一个小问题。

make[3]: Entering directory
`/home/paul/openwrt-15.05/package/firmware/wireless-regdb'
mkdir -p /home/paul/openwrt-15.05/dl
--2020-03-15 20:26:20--
http://mirror2.openwrt.org/sources/wireless-regdb-2017-10-20.tar.xz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80...
connected.
HTTP request sent, awaiting response... 404 Not Found
2020-03-15 20:26:21 ERROR 404: Not Found.

Download failed.
--2020-03-15 20:26:21--
http://downloads.openwrt.org/sources/wireless-regdb-2017-10-20.tar.xz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 176.9.48.73
Connecting to downloads.openwrt.org (downloads.openwrt.org)|176.9.48.73|:80...
connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://sources.openwrt.org/wireless-regdb-2017-10-20.tar.xz
[following]
--2020-03-15 20:26:22--
http://sources.openwrt.org/wireless-regdb-2017-10-20.tar.xz
Resolving sources.openwrt.org (sources.openwrt.org)... 148.251.151.136
Connecting to sources.openwrt.org (sources.openwrt.org)|148.251.151.136|:80...
connected.
HTTP request sent, awaiting response... 404 Not Found
2020-03-15 20:26:23 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
Checking out files from the git repository...
Cloning into 'wireless-regdb-2017-10-20'...
error: RPC failed; result=35, HTTP code = 0
fatal: The remote end hung up unexpectedly
make[3]: *** [/home/paul/openwrt-15.05/dl/wireless-regdb-2017-10-20.tar.xz]
Error 128
make[3]: Leaving directory
`/home/paul/openwrt-15.05/package/firmware/wireless-regdb'
make[2]: *** [package/firmware/wireless-regdb/compile] Error 2
make[2]: Leaving directory `/home/paul/openwrt-15.05'
make[1]: ***
[/home/paul/openwrt-15.05/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile]
Error 2
make[1]: Leaving directory `/home/paul/openwrt-15.05'
make: *** [world] Error 2

 我去找了一下，openwrt上确实已经不存在这个文件了，又去
http://ftp.iij.ad.jp/pub/linux/kernel/software/network/wireless-regdb/
找了，里面已经没有wireless-regdb-2017-10-20.tar.xz。
网上最多的就是wireless-regdb-2017-10-20-4343d359.tar.xz，我下载了一个，改名以后发现出现新的报错：

make[3]: Entering directory
`/home/paul/openwrt-15.05/package/firmware/wireless-regdb'
. /home/paul/openwrt-15.05/include/shell.sh; xzcat
/home/paul/openwrt-15.05/dl/wireless-regdb-2017-10-20.tar.xz | tar -C
/home/paul/openwrt-15.05/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/wireless-regdb-2017-10-20/..
-xf -

Applying
./patches/100-regdb-write-firmware-file-format-version-code-20.patch using
plaintext:
can't find file to patch at input line 13
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From: Johannes Berg <johannes.berg@intel.com>
|Date: Mon, 9 Oct 2017 11:50:57 +0200
|Subject: [PATCH] regdb: write firmware file format (version code 20)
|
|TODO: clean up the Makefile stuff ...
|
|Signed-off-by: Johannes Berg <johannes.berg@intel.com>
|---
| create mode 100755 db2fw.py
|
|--- a/Makefile
|+++ b/Makefile
--------------------------
No file to patch.  Skipping patch.
3 out of 3 hunks ignored
can't find file to patch at input line 105
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/README
|+++ b/README
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
patching file db2fw.py
Patch failed!  Please fix
./patches/100-regdb-write-firmware-file-format-version-code-20.patch!
make[3]: ***
[/home/paul/openwrt-15.05/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/wireless-regdb-2017-10-20/.prepared_7a540624d40882d2dba0397dfcef35f0]
Error 1
make[3]: Leaving directory
`/home/paul/openwrt-15.05/package/firmware/wireless-regdb'
make[2]: *** [package/firmware/wireless-regdb/compile] Error 2
make[2]: Leaving directory `/home/paul/openwrt-15.05'
make[1]: ***
[/home/paul/openwrt-15.05/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile]
Error 2
make[1]: Leaving directory `/home/paul/openwrt-15.05'
make: *** [world] Error 2

PaulYoung0265, 2020-03-21:
还有后继，真是对不住。
更新代码后重新编译，到下载v2ray-core-4.22.1.tar.gz时提示校验不通过，然后其他的下载链接都不对，无法下载到。
我验证了一下，下载链接和project V给出的的确不一致无法下载。
手动从project V给的GitHub链接上下载了一个source code放进dl目录，编译成功，也成功刷了进去。
[image: image.png]
在passwall里面，trojan节点可以使用，但一切换到v2ray节点，就会显示tcp udp未运行。
[image: image.png]

去自动更新设置里面一看，居然没显示版本号，手动更新也不行。
[image: image.png]
SSH进去，未发现v2ray的进程，/etc/passwall/bin下面也没有v2ray的软链
[image: image.png]

另外，LED灯始终是红色的，没有启动闪烁，正常运行也不转变成blue...

感觉曙光就在眼前啦，麻烦大神给诊断一下

PaulYoung0265, 2020-03-22:
早上看见有代码更新，又编译了一次，这次卡在 openssl 150的patch上了， ERROR LOG见下：

Applying ./patches/150-openssl.cnf-add-engines-conf.patch using plaintext:
patching file apps/openssl.cnf
patch: **** malformed patch at line 4: @@ -22,6 +22,82 @@ oid_section
    = new_oids

Patch failed!  Please fix ./patches/150-openssl.cnf-add-engines-conf.patch!
make[3]: ***
[/home/paul/openwrt-15.05/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/openssl-1.1.1d/.prepared_628ad39c1ecac05c3f4bdcb108df0860]
Error 1
make[3]: Leaving directory `/home/paul/openwrt-15.05/package/libs/openssl'
make[2]: *** [package/libs/openssl/compile] Error 2
make[2]: Leaving directory `/home/paul/openwrt-15.05'
make[1]: ***
[/home/paul/openwrt-15.05/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile]
Error 2
make[1]: Leaving directory `/home/paul/openwrt-15.05'
make: *** [world] Error 2

PaulYoung0265, 2020-03-22:
咦，怎么又卡v2ray这里了-----后面那一堆downlaod failed 可以修复吗？

make[3]: Entering directory
`/home/paul/openwrt-15.05/feeds/lienol/package/v2ray'
mkdir -p /home/paul/openwrt-15.05/dl
/home/paul/openwrt-15.05/scripts/download.pl "/home/paul/openwrt-15.05/dl"
"v2ray-core-4.22.1.tar.gz"
"31c1934eeac3552c7ab68eac9dc3e964e05f3c743b3733b0b6a0159c499d6" "" "
https://codeload.github.com/v2ray/v2ray-core/tar.gz/v4.22.1?"
--2020-03-22 18:54:20--
https://codeload.github.com/v2ray/v2ray-core/tar.gz/v4.22.1?/v2ray-core-4.22.1.tar.gz
Resolving codeload.github.com (codeload.github.com)... 140.82.113.9
Connecting to codeload.github.com (codeload.github.com)|140.82.113.9|:443...
connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: 'STDOUT'

    [            <=>
                                                     ] 6,835,327   2.79MB/s
  in 2.3s

2020-03-22 18:54:25 (2.79 MB/s) - written to stdout [6835327]

MD5 sum of the downloaded file does not match (file:
413e413e748c58250ba7db61a102e525, requested:
31c1934eeac3552c7ab68eac9dc3e964e05f3c743b3733b0b6a0159c495019d- deleting
download.
--2020-03-22 18:54:25--
http://mirror2.openwrt.org/sources/v2ray-core-4.22.1.tar.gz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80...
connected.
HTTP request sent, awaiting response... 404 Not Found
2020-03-22 18:54:26 ERROR 404: Not Found.

Download failed.
--2020-03-22 18:54:26--
http://downloads.openwrt.org/sources/v2ray-core-4.22.1.tar.gz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 176.9.48.73
Connecting to downloads.openwrt.org (downloads.openwrt.org)|176.9.48.73|:80...
connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://sources.openwrt.org/v2ray-core-4.22.1.tar.gz [following]
--2020-03-22 18:54:27--  http://sources.openwrt.org/v2ray-core-4.22.1.tar.gz
Resolving sources.openwrt.org (sources.openwrt.org)... 148.251.151.136
Connecting to sources.openwrt.org (sources.openwrt.org)|148.251.151.136|:80...
connected.
HTTP request sent, awaiting response... 404 Not Found
2020-03-22 18:54:28 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[3]: *** [/home/paul/openwrt-15.05/dl/v2ray-core-4.22.1.tar.gz] Error 2
make[3]: Leaving directory
`/home/paul/openwrt-15.05/feeds/lienol/package/v2ray'
make[2]: *** [package/feeds/lienol/v2ray/compile] Error 2
make[2]: Leaving directory `/home/paul/openwrt-15.05'
make[1]: ***
[/home/paul/openwrt-15.05/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile]
Error 2
make[1]: Leaving directory `/home/paul/openwrt-15.05'
make: *** [world] Error 2

PaulYoung0265, 2020-03-22:
先手动下载放进去继续，看看还有什么其他问题，果然有，卡死在这里了：
make[3]: Leaving directory `/home/paul/openwrt-15.05/package/libs/polarssl'
make[3]: Entering directory
`/home/paul/openwrt-15.05/package/libs/ustream-ssl'
mkdir -p /home/paul/openwrt-15.05/dl
echo "Checking out files from the git repository..."; mkdir -p
/home/paul/openwrt-15.05/tmp/dl && cd /home/paul/openwrt-15.05/tmp/dl && rm
-rf ustream-ssl-2015-07-09 && [ \! -d ustream-ssl-2015-07-09 ] && git clone
git://nbd.name/ustream-ssl.git ustream-ssl-2015-07-09 --recursive && (cd
ustream-ssl-2015-07-09 && git checkout
c2d73c22618e8ee444e8d346695eca908ecb72d3 && git submodule update) && echo
"Packing checkout..." && rm -rf ustream-ssl-2015-07-09/.git &&   tar czf
/home/paul/openwrt-15.05/tmp/dl/ustream-ssl-2015-07-09-c2d73c22618e8ee444e8d346695eca908ecb72d3.tar.gz
ustream-ssl-2015-07-09 && mv
/home/paul/openwrt-15.05/tmp/dl/ustream-ssl-2015-07-09-c2d73c22618e8ee444e8d346695eca908ecb72d3.tar.gz
/home/paul/openwrt-15.05/dl/ && rm -rf ustream-ssl-2015-07-09;
Checking out files from the git repository...
Cloning into 'ustream-ssl-2015-07-09'...
[image: image.png]

PaulYoung0265, 2020-03-25:
看见代码更新，V2RAY也更新到了4.23.1，兴冲冲地来编译。
但是没想到，现实很快就给了我一棍子，还是老问题。
v2ray下载报MD5校验错误，其他下载地址不可用---手动下载的版本进去编译能过，实际不能用，passwall里面TCP一直灰色起不来。
这个下载链接的问题已经很多次了，就不能取消这个MD5验证吗？
我一直很感激你们的努力维护，也希望我的小小建议能对你们有帮助。

PaulYoung0265, 2020-03-26:
mkdir -p /home/paul/openwrt-15.05/dl
/home/paul/openwrt-15.05/scripts/download.pl "/home/paul/openwrt-15.05/dl"
"v2ray-core-4.23.1.tar.gz"
"474b3aeed069d9867f7603a0544abcc0f31386cef9254423577ab752fc8d4dcc" "" "
https://codeload.github.com/v2ray/v2ray-core/tar.gz/v4.23.1?"
--2020-03-26 10:54:41--
https://codeload.github.com/v2ray/v2ray-core/tar.gz/v4.23.1?/v2ray-core-4.23.1.tar.gz
Resolving codeload.github.com (codeload.github.com)... 192.30.255.121
Connecting to codeload.github.com (codeload.github.com)|192.30.255.121|:443...
connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: 'STDOUT'

    [          <=>
                                                  ] 6,990,943   3.51MB/s
in 1.9s

2020-03-26 10:54:45 (3.51 MB/s) - written to stdout [6990943]

MD5 sum of the downloaded file does not match (file:
026720369632c0bbfeb1fa079e5529f9, requested:
474b3aeed069d9867f7603a0544abcc0f31386cef9254423577ab752fc8d4dcc) -
deleting download.
--2020-03-26 10:54:45--
http://mirror2.openwrt.org/sources/v2ray-core-4.23.1.tar.gz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80...
connected.
HTTP request sent, awaiting response... 404 Not Found
2020-03-26 10:54:47 ERROR 404: Not Found.

Download failed.
--2020-03-26 10:54:47--
http://downloads.openwrt.org/sources/v2ray-core-4.23.1.tar.gz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 176.9.48.73
Connecting to downloads.openwrt.org (downloads.openwrt.org)|176.9.48.73|:80...
connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://sources.openwrt.org/v2ray-core-4.23.1.tar.gz [following]
--2020-03-26 10:54:48--  http://sources.openwrt.org/v2ray-core-4.23.1.tar.gz
Resolving sources.openwrt.org (sources.openwrt.org)... 148.251.151.136
Connecting to sources.openwrt.org (sources.openwrt.org)|148.251.151.136|:80...
connected.
HTTP request sent, awaiting response... 404 Not Found
2020-03-26 10:54:49 ERROR 404: Not Found.

Download failed.