Lienol/openwrt #37: dev-19.07  passwall 无法订阅更新

prowjy, 2020-02-16:
2020-02-16 21:28:14: 删除所有防火墙规则...
2020-02-16 21:28:27: 清除定时执行命令。
2020-02-16 21:28:27: 关闭相关程序，清理相关文件和缓存完成。
2020-02-16 21:28:37: 开始订阅...
--2020-02-16 21:28:37--  https://github.com/ssrsub/ssr/raw/master/v2ray
Resolving github.com... 13.250.177.223
Connecting to github.com|13.250.177.223|:443... failed: Operation timed out.
Retrying.

--2020-02-16 21:28:48--  (try: 2)  https://github.com/ssrsub/ssr/raw/master/v2ray
Connecting to github.com|13.250.177.223|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/ssrsub/ssr/master/v2ray [following]
--2020-02-16 21:28:55--  https://raw.githubusercontent.com/ssrsub/ssr/master/v2ray
Resolving raw.githubusercontent.com... 151.101.228.133
Connecting to raw.githubusercontent.com|151.101.228.133|:443... failed: Connection refused.
--2020-02-16 21:28:55--  https://raw.githubusercontent.com/voken100g/AutoSSR/master/recent
Resolving raw.githubusercontent.com... 151.101.228.133
Connecting to raw.githubusercontent.com|151.101.228.133|:443... failed: Connection refused.
2020-02-16 21:28:55: /usr/share/passwall/subscribe.lua:292: node result is empty
2020-02-16 21:28:55: stack traceback:
	/usr/share/passwall/subscribe.lua:449: in function </usr/share/passwall/subscribe.lua:447>
	[C]: in function 'assert'
	/usr/share/passwall/subscribe.lua:292: in function 'update_node'
	/usr/share/passwall/subscribe.lua:438: in function </usr/share/passwall/subscribe.lua:422>
	[C]: in function 'xpcall'
	/usr/share/passwall/subscribe.lua:447: in main chunk
	[C]: ?
2020-02-16 21:28:55: 发生错误, 正在恢复服务
2020-02-16 21:28:55: 订阅完毕...

prowjy, 2020-02-16:
.config  用的是之前编译的   是不是要重新生成一下

prowjy, 2020-02-16:
重新编译  还是不行

prowjy, 2020-02-16:
重新编译还是不行

prowjy, 2020-02-16:
无法启动  开启后TCP服务闪一下

iamsxm, 2020-02-17:
看日志是你的网络超时了，你先自己手动导入链接，然后勾选上使用代理更新

luckyyyyy, 2020-02-17:
网络问题，看日志。

closed 2020-02-18 by Lienol