Lienol/openwrt #38: Collected errors:  * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/releases/19.07-SNAPSHOT/packages/x86_64/lienol/Packages.gz, wget returned 8.

chenxiccc, 2020-02-16:
更新软件包列表时报错：

Collected errors:
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/releases/19.07-SNAPSHOT/packages/x86_64/lienol/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/releases/19.07-SNAPSHOT/packages/x86_64/darkmatter/Packages.gz, wget returned 8.

sqliuchang, 2020-02-16:
这个两个源是如何产生的我不懂。。。
只能在这里看到是把官方源替换成了中科大的源
https://github.com/Lienol/openwrt/blob/4392beb45b5b0261dc3b92341678bcf1d2dd9685/package/default-settings/files/zzz-default-settings#L15
但是我知道去/etc/opkg/distfeeds.conf 删除这两行就行了

closed 2020-02-18 by Lienol

CuminLo, 2020-03-02:
刚刚编译完成的固件使用 opkg upgrade xx 也出现了这样的问题

iMaxson, 2020-03-11:
我也是遇到这个问题，有办法在编译前解决吗？？

jbao69, 2020-04-05:
Same problem: 
 Failed to download https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/lienol/Packages.gz, wget returned 8.
Can you help to solve the issue?

Yohoa, 2020-06-17:
遇到同样的问题，请问大家有什么建议吗？