Lienol/openwrt #40: 2/16 dev-master 固件 TROJAN 订阅链接找不到 prot

ATErBion, 2020-02-17:
2/16 dev-master 固件

单独启用 TROJAN 链接订阅直接报错，如果 VMESS 链接和 TROJAN 链接同时启用，订阅时无法添加任何节点，只能单独启用 VMESS 链接。

看日志是 TROJAN 链接找不到 prot 导致出错

还是给出 TROJAN 链接所使用格式吧，subscription.sh 看了下不知道怎么改
```
trojan://密码@节点IP:端口_allowInsecure=1&peer=节点域名#节点备注

#订阅文件Base64解码后，trojan链接#号后节点备注还需要 URLEncode
```
> 两个订阅链接在 IOS 小火箭下都使用正常
> VMESS 链接内有10个节点，6个 v2ray 节点，4个 ssr 节点纯做提示信息用；
> TROJAN 链接内有10个节点，6个 trojan 节点，4个 ssr 节点纯做提示信息用。
> 现在已经暂时刷回 2/15 的版本使用

**2/16 固件版本的订阅日志**
#以下日志内域名和 ip 均替换成 XXX
```
2020-02-16 16:49:35: 开始订阅...
--2020-02-16 16:49:35--  https://XXX.XXX/rss/XXXXXXXXXXXX?net_type=VMESS
Resolving XXX.XXX... XXX.XXX.XXX.XXX, XXX.XXX.XXX.XXX
Connecting to XXX.XXX|XXX.XXX.XXX.XXX|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/plain]
Saving to: 'STDOUT'

     0K ..                                                     3.24K=0.9s

2020-02-16 16:49:37 (3.24 KB/s) - written to stdout [2876]

2020-02-16 16:49:37: 成功解析节点数量: 10
2020-02-16 16:49:37: 新增节点数量: 10 删除节点数量: 0
2020-02-16 16:49:37: 订阅完毕...
2020-02-16 16:49:38: 删除所有防火墙规则...
2020-02-16 16:49:38: 清除定时执行命令。
2020-02-16 16:49:38: 关闭相关程序，清理相关文件和缓存完成。
2020-02-16 16:49:40: TCP节点：日本·1000M·空闲，节点：XXX.XXX.XXX.XXX:XXXX，监听端口：XXXX
2020-02-16 16:49:40: DNS：pdnsd...
2020-02-16 16:49:40: 开始加载防火墙规则...
2020-02-16 16:49:40: 默认代理模式：大陆白名单
2020-02-16 16:49:40: IPv4 防火墙TCP转发规则加载完成！
2020-02-16 16:49:40: 添加DNS劫持规则...
2020-02-16 16:49:40: dnsmasq：生成配置文件并重启服务。
2020-02-16 16:49:40: 已启动守护进程。
2020-02-16 16:49:40: 设置自动更新规则在每天 1 点。
2020-02-16 16:49:40: 设置节点订阅自动更新规则在每天 0 点。
2020-02-16 16:49:40: 运行完成！

2020-02-16 16:50:44: 开始订阅...
--2020-02-16 16:50:44--  https://XXX.XXX/rss/XXXXXXXXXXXX?net_type=TROJAN
Resolving XXX.XXX... XXX.XXX.XXX.XXX, XXX.XXX.XXX.XXX
Connecting to XXX.XXX|XXX.XXX.XXX.XXX|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/plain]
Saving to: 'STDOUT'

     0K ..                                                     8.65M=0s

2020-02-16 16:50:45 (8.65 MB/s) - written to stdout [2286]

2020-02-16 16:50:45: /usr/share/passwall/subscribe.lua:225: attempt to concatenate field 'port' (a nil value)
2020-02-16 16:50:45: stack traceback:
	/usr/share/passwall/subscribe.lua:449: in function </usr/share/passwall/subscribe.lua:447>
	/usr/share/passwall/subscribe.lua:225: in function 'processData'
	/usr/share/passwall/subscribe.lua:394: in function 'parse_link'
	/usr/share/passwall/subscribe.lua:432: in function 'callback'
	/usr/lib/lua/luci/model/uci.lua:281: in function 'foreach'
	/usr/share/passwall/subscribe.lua:425: in function </usr/share/passwall/subscribe.lua:422>
	[C]: in function 'xpcall'
	/usr/share/passwall/subscribe.lua:447: in main chunk
	[C]: ?
2020-02-16 16:50:45: 发生错误, 正在恢复服务
2020-02-16 16:50:45: 订阅完毕...

```

**2/15 固件版本的订阅日志**
OpenWrt SNAPSHOT / LuCI Master-Lienol git-20.045.46446-a52bf8b 

VMESS 链接和 TROJAN 链接同时启用没有问题，TROJAN 链接能认出来一个节点（可正常用，虽然节点备注没认出来）
```
2020-02-16 17:07:47: 删除所有防火墙规则...
2020-02-16 17:07:47: 清除定时执行命令。
2020-02-16 17:07:47: 关闭相关程序，清理相关文件和缓存完成。
2020-02-16 17:07:48: 没有选择节点！
2020-02-16 17:07:47: 在线订阅节点已全部删除
2020-02-16 17:08:00: 开始订阅...
2020-02-16 17:08:00: 正在订阅：https://XXX.XXX/rss/XXXXXXXXXXXX?net_type=VMESS
2020-02-16 17:08:00: 西部世界v： V2ray节点新增 6 个，修改 0 个，删除 0 个。
2020-02-16 17:08:00: 正在订阅：https://XXX.XXX/rss/XXXXXXXXXXXX?net_type=TROJAN
2020-02-16 17:08:00: 西部世界t： Trojan节点新增 1 个，修改 0 个，删除 0 个。
2020-02-16 17:08:00: 订阅完毕...
```

Lienol, 2020-02-17:
订阅链接发我邮箱测试

ATErBion, 2020-02-18:
> 订阅链接发我邮箱测试

请查收

closed 2020-02-18 by Lienol

colusiii, 2020-03-13:
面临同样的问题，也是西部世界的trojan订阅，请问大佬有啥解决方案么，而且今天手动录入西部世界的trojan节点在passwall里也无法穿越成功。