Lienol/openwrt #43: 2020.02.17-10:52 Lienol/openwrt/dev-master无法订阅节点

heroareslx, 2020-02-17:
昨天16版的也无法订阅节点，但是日志无显示，今天版本日志的显示如下：

2020-02-17 14:45:15: 开始订阅...
2020-02-17 14:45:17: /usr/share/passwall/subscribe.lua:237: attempt to concatenate field 'port' (a nil value)
2020-02-17 14:45:17: stack traceback:
	/usr/share/passwall/subscribe.lua:456: in function </usr/share/passwall/subscribe.lua:454>
	/usr/share/passwall/subscribe.lua:237: in function 'processData'
	/usr/share/passwall/subscribe.lua:401: in function 'parse_link'
	/usr/share/passwall/subscribe.lua:439: in function 'callback'
	/usr/lib/lua/luci/model/uci.lua:281: in function 'foreach'
	/usr/share/passwall/subscribe.lua:432: in function </usr/share/passwall/subscribe.lua:429>
	[C]: in function 'xpcall'
	/usr/share/passwall/subscribe.lua:454: in main chunk
	[C]: ?
2020-02-17 14:45:17: 发生错误, 正在恢复服务
2020-02-17 14:45:17: 订阅完毕...

ATErBion, 2020-02-17:
什么节点？看看跟我那个问题是不是类似？

heroareslx, 2020-02-17:
> 什么节点？看看跟我那个问题是不是类似？

就是正常的通用订阅节点，https的

iamsxm, 2020-02-17:
看样子是获取的信息端口port为空值

heroareslx, 2020-02-17:
> 看样子是获取的信息端口port为空值

相同的节点在以前版本的固件里订阅都是没问题的，这两个版本都订阅不了，很奇怪

ATErBion, 2020-02-17:
> > 什么节点？看看跟我那个问题是不是类似？
> 
> 就是正常的通用订阅节点，https的

你可以先试试纯vmess的订阅，通用不知道是不是含Trojan

Lienol, 2020-02-17:
订阅链接发我邮箱测试

closed 2020-02-18 by Lienol