Lienol/openwrt #57: 求助！make download 时 提示：

zgs360, 2020-02-19:
[image]
[image]

编译时出错

zgs360, 2020-02-19:
求大神解决！

Lienol, 2020-02-19:
你make menuconfig了吗？

zgs360, 2020-02-20:
@Lienol 我make menuconfig 后，在make -j1 V=s 之前，先下载，因为前面直接编译时，错误而中断，查找原因，重新clone，还是不行，后来在lede 中加入passwall 中编译成功，但功能tcp什么不显示链接成功，还是用大神的源码编译，一直无法成功，所以才用 make download V=s ,就上面两个提示。我还查找是不是ubuntu14.04 的gcc 版本你的原因，我又更新了gcc6，还不是行。
编译提示错误是下面的图。
[image]

zgs360, 2020-02-20:
@Lienol 我的编译环境：ubuntu14.04，我选择的86X64

closed 2020-03-08 by Lienol