Lienol/openwrt #67: 网易云 问题 编译成功 打开后显示

chiyu521, 2020-02-21:
Failed to execute cbi dispatcher target for entry '/admin/services/unblockneteasemusic/general'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/unblockneteasemusic.lua:1: module 'luci.fs' not found:
	no field package.preload['luci.fs']
	no file './luci/fs.lua'
	no file '/usr/share/lua/luci/fs.lua'
	no file '/usr/share/lua/luci/fs/init.lua'
	no file '/usr/lib/lua/luci/fs.lua'
	no file '/usr/lib/lua/luci/fs/init.lua'
	no file './luci/fs.so'
	no file '/usr/lib/lua/luci/fs.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/model/cbi/unblockneteasemusic.lua:1: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:845: in function </usr/lib/lua/luci/dispatcher.lua:839>

iamsxm, 2020-02-21:
参考我的[解决方法](https://github.com/Lienol/openwrt/issues/52)

chiyu521, 2020-02-21:
感谢 楼上 我还遇到一个问题 就是 设置完passwall后 NAT就无法变成1了  不知道有没有解决方法

closed 2020-02-26 by Lienol