Lienol/openwrt #83: 编译固件 uClibc 报错问题

tzc85692548, 2020-02-23:
/home/joseph/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.5.0/../../../../x86_64-openwrt-linux-musl/bin/ar: array_type_info.o: No such file or directory
make[4]: *** [src/libuClibc++.a] Error 1
make[4]: Leaving directory `/home/joseph/openwrt/build_dir/target-x86_64_musl/uClibc++-0.2.5'
make[3]: *** [/home/joseph/openwrt/build_dir/target-x86_64_musl/uClibc++-0.2.5/.built] Error 2
make[3]: Leaving directory `/home/joseph/openwrt/package/libs/uclibc++'
time: package/libs/uclibc++/compile#3.88#0.60#5.36
make[2]: *** [package/libs/uclibc++/compile] Error 2
make[2]: Leaving directory `/home/joseph/openwrt'
make[1]: *** [/home/joseph/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/joseph/openwrt'
make: *** [world] Error 2

 Lienol 大神好，今天在编译您固件时遇到这么一段报错，编译不过去。麻烦帮忙看一下，谢谢！

我自己试过make clean然后重新编译；也试过重新git clone，还是一样的报错。

zgs360, 2020-02-28:
解决了么？我也是同样的错误！

zgs360, 2020-02-28:
@tzc85692548 解决了么？我也是同样的错误！

tzc85692548, 2020-03-11:
大家找到解决办法了吗？

ClayMoreBoy, 2020-03-15:
同样这个错误

txdywy, 2020-07-23:
如何解决...