Lienol/openwrt #86: dev-19.07和dev-master编译时，go1.13.7.src.tar.gz无法下载，action和本地都是一样。

EthanChoy, 2020-02-24:
Hash of the downloaded file does not match (file: e4ad42cc5f5c19521fbbbde3680995f2546110b5c6aa2b48c3754ff7af9b41f4, requested: b13bf04633d4d8cf53226ebeaace8d4d2fd07ae6fa676d0844a688339debec34) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/go1.13.7.src.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
Makefile:287: recipe for target '/home/runner/work/openwrt-actions/openwrt-actions/openwrt/dl/go1.13.7.src.tar.gz' failed
make[3]: Leaving directory '/home/runner/work/openwrt-actions/openwrt-actions/openwrt/feeds/packages/lang/golang/golang'
make[3]: *** [/home/runner/work/openwrt-actions/openwrt-actions/openwrt/dl/go1.13.7.src.tar.gz] Error 2
time: package/feeds/packages/golang/host-compile#1.28#1.15#621.37
package/Makefile:111: recipe for target 'package/feeds/packages/golang/host/compile' failed
make[2]: *** [package/feeds/packages/golang/host/compile] Error 2
make[2]: Leaving directory '/home/runner/work/openwrt-actions/openwrt-actions/openwrt'
package/Makefile:107: recipe for target '/home/runner/work/openwrt-actions/openwrt-actions/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/home/runner/work/openwrt-actions/openwrt-actions/openwrt'
make[1]: *** [/home/runner/work/openwrt-actions/openwrt-actions/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make: *** [world] Error 2
/home/runner/work/openwrt-actions/openwrt-actions/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
##[error]Process completed with exit code 2.

nichwang88, 2020-02-24:
我在编译dev-19.07的时候也出现类似的错误，编译几次了，没有一次成功，已经出墙全局模式了。烦请L大帮忙看看什么原因？

```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 * 	luci-base
 * opkg_install_cmd: Cannot install package default-settings.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-lib-docker:
 * 	luci-lib-json
 * opkg_install_cmd: Cannot install package luci-lib-docker.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for autocore:
 * 	bc
 * 	lm-sensors
 * opkg_install_cmd: Cannot install package autocore.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
 * 	kmod-fs-exfat
 * 	ntfs-3g
 * opkg_install_cmd: Cannot install package automount.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for autosamba:
 * 	luci-app-samba
 * opkg_install_cmd: Cannot install package autosamba.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ddns-scripts_aliyun:
 * 	ddns-scripts
 * opkg_install_cmd: Cannot install package ddns-scripts_aliyun.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ddns-scripts_dnspod:
 * 	ddns-scripts
 * opkg_install_cmd: Cannot install package ddns-scripts_dnspod.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-adbyby-plus:
 * 	coreutils
 * 	coreutils-nohup
 * opkg_install_cmd: Cannot install package luci-app-adbyby-plus.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-filetransfer:
 * 	luci
 * 	luci-lib-nixio
 * opkg_install_cmd: Cannot install package luci-app-filetransfer.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-flowoffload:
 * 	pdnsd-alt
 * opkg_install_cmd: Cannot install package luci-app-flowoffload.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-usb-printer:
 * 	p910nd
 * opkg_install_cmd: Cannot install package luci-app-usb-printer.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/nicholas/openwrt'
package/Makefile:108: recipe for target '/home/nicholas/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/nicholas/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/nicholas/openwrt'
/home/nicholas/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

xnxy2012, 2020-02-24:
好像是下载的文件的md5值设置有错误

sypopo, 2020-02-24:
我选了UnblockNeteaseMusicGo就编译失败

nichwang88, 2020-02-24:
不知何故，我刚刚又编译一次居然成功了，也没搞懂怎么出的问题，也没搞懂怎么就又好了。

Lienol, 2020-02-25:
已更新
请更新feeds重新编译即可

closed 2020-02-25 by Lienol