Lienol/openwrt #89: 2月24日DEV版本“节点列表”不显示PING值

zk703, 2020-02-24:
This repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins or for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.

Lienol, 2020-02-25:
正常的
输入 rm -rf /tmp/luci-*
试试

zk703, 2020-02-25:
收到，昨天夜里试了好多次都不行，而且切换passwall的时候非常慢，在节点列表里面直接点击使用节点反应也非常慢，今天上午起来后再试一下，ping值有了，暂时还没有卡顿的现象，可能是个例，麻烦你了！ 发送自 Windows 10 版邮件应用 发件人: Lienol发送时间: 2020年2月25日 13:51收件人: Lienol/openwrt抄送: zk703; Author主题: Re: [Lienol/openwrt] 2月24日DEV版本“节点列表”不显示PING值 (#89) 正常的输入 rm -rf /tmp/luci-*试试

closed 2020-02-25 by Lienol