Lienol/openwrt #102: 端口映射upnp貌似有问题

WROIATE, 2020-02-26:
在2020.1.23自编译版本中存在该问题，表现为映射内网ftp端口无法连接以及使用被动模式，而这在lean的固件中无问题

loarland, 2020-02-29:
Sat Feb 29 11:06:50 2020 daemon.err miniupnpd[13794]: Failed to add NAT-PMP 16217 tcp->10.10.10.9:3000 'NAT-PMP 16217 tcp'
我日志里面全是这个报错

loarland, 2020-02-29:
> > 那你看看现在几号了…
> 
> 昨天编译实测并未修复，请不要意识流误导
> [image]

爬资料貌似解决问题了，/etc/config/upnpd这个文件config里面加入一行
option external_iface 'wan'
然后重启upnp服务就可以了

chenxiccc, 2020-03-04:
Followings, same problem.
And new added port forwards won't work unless i reboot openwrt.

kinkitchen, 2020-03-19:
> > > 那你看看现在几号了…
> > 
> > 
> > 昨天编译实测并未修复，请不要意识流误导
> > [image]
> 
> 爬资料貌似解决问题了，/etc/config/upnpd这个文件config里面加入一行
> option external_iface 'wan'
> 然后重启upnp服务就可以了

昨晚编译的，UPNP也是不行，失效的。尝试增加/etc/config/upnpd这个文件config里面加入一行
option external_iface 'wan'，也是失败。

Lienol, 2020-03-27:
要么回滚旧版upnp算了

WROIATE, 2020-03-28:
> 要么回滚旧版upnp算了

不太清楚新版旧版的区别，但如果旧版是正常的那的确不如回滚旧版

closed 2020-05-16 by Lienol