Lienol/openwrt #110: 【19.07平稳分支】27日后面的固件，passwall的BUG

ilee2go, 2020-02-28:
Li大，我之前在电报群里有反馈过，群里消息太多，不知道你能不能听懂我的表述。
我记得你之前有说固件升级加入了刷新缓存，不需要再背诵：ipconfig /flushdns了，好像就是27日那天起吧，然后就有了现在的问题。

比如现在passwall是正常的，代理模式是GFW名单，所有gfwlist里面的网站均可正常访问。
然后我现在打开：www.google.com/ncr，可以正常访问。
接着，我在不改变任何设置的情况下，我就点了一次passwall里面的保存并应用，然后发现谷歌打不开了，但是其它的网站，如：www.facebook，可以正常打开！
再来一次一样的步骤，我再点一次passwall里面的保存后，facebook网站也打不开了，但是其它的网站，如：www.twitter.com，可以正常打开。

意思就是，每点一次passwall的保存，就会造成之前能开的网站，无法打开，之前没开过的网站，依旧可以正常通过代理访问！

但是这样的情况在25日之前的版本是不存在的，之前的版本没改什么配置，不管怎么点保存，都没任何影响。只不过是黑名单添加了新域名不会即时生效而已，并不会影响之前本来就可以打开的网站。

如果有需要，我可以提供远程给你复现下问题。

[image]

luan5233, 2020-02-28:
我也有类似的问题，大概就是GFW模式，我的google和youtube打开总有问题，打不开

ilee2go, 2020-02-29:
> 我也有类似的问题，大概就是GFW模式，我的google和youtube打开总有问题，打不开

今天的新固件问题依旧，点了保存，之前的网站就打不开了，必需刷新一次本机DNS，过段时间后才能打开。

ilee2go, 2020-03-05:
@Lienol 
实在不愿多打扰，但是还是没找到问题。
群里一直都说，从几天前开始，使用Lienol/openwrt源码任意分支编译的固件。切换线路或者代理模式后都不再必须要flushdns了！
但是我实际测试的情况，就是上面的那样，点了保存，之前打开的网站就失效无法打开，必需刷新才能打开。

wefuxkgfw, 2020-03-05:
> @Lienol
> 实在不愿多打扰，但是还是没找到问题。
> 群里一直都说，从几天前开始，使用Lienol/openwrt源码任意分支编译的固件。切换线路或者代理模式后都不再必须要flushdns了！
> 但是我实际测试的情况，就是上面的那样，点了保存，之前打开的网站就失效无法打开，必需刷新才能打开。

GFW默认模式下会这样。。要不先白名单顶一下吧。。。

Lienol, 2020-03-05:
更新最新源码试试？

ilee2go, 2020-03-06:
经测试，2020.03.06的固件已经解决此问题，感谢Li大！

closed 2020-03-06 by ilee2go