Lienol/openwrt #112: 大神给看看这K3啥问题，是不是Readme解释的有问题

taoqixfz, 2020-02-28:
我只选择了型号，竟然不能成功编译
Collected errors:
 * check_data_file_clashes: Package k3wifi wants to install file /home/lf/test/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/lib/firmware/brcm/brcmfmac4366c-pcie.bin
	But that file is already provided by package  * brcmfmac-firmware-4366c0-pcie
 * opkg_install_cmd: Cannot install package k3wifi.
 * resolve_conffiles: Existing conffile /home/lf/test/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /home/lf/test/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/ppp/options-opkg.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/lf/test/openwrt'
package/Makefile:108: recipe for target '/home/lf/test/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_install' failed
make[1]: *** [/home/lf/test/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/lf/test/openwrt'
/home/lf/test/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

Lienol, 2020-02-29:
编译时把brcmfmac-firmware-4366c0-pcie去掉
再勾选k3wifi试试

taoqixfz, 2020-02-29:
> 编译时把brcmfmac-firmware-4366c0-pcie
> 去掉再重新插入k3wifi试试

我把k3wifi去掉了，编译好后刷入路由器发现没有5G信号，编译时把brcmfmac-firmware-4366c0-pcie去掉的选项在什么地方

closed 2020-03-07 by Lienol

taoqixfz, 2020-03-08:
已成功编译，感谢