Lienol/openwrt #113: 请教大家一个编译出错问题

harry3633, 2020-02-29:
搞了好几天了，也没搜索到，请教大家一下这是什么问题
```
/home/harry/openwrt/staging_dir/host/bin/otrx create /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tenda-ac9-squashfs.trx.new -f /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tenda-ac9-kernel.bin -a 1024 -f /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tenda-ac9-squashfs.trx -a 0x10000 -A /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/fs_mark
mv /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tenda-ac9-squashfs.trx.new /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tenda-ac9-squashfs.trx
cp /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tenda-ac9-squashfs.trx /home/harry/openwrt/bin/targets/bcm53xx/generic/openwrt-bcm53xx-tenda-ac9-squashfs.trx
rm -f /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c5-v2-kernel.bin
cp /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/zImage /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c5-v2-kernel.bin
cat /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/image-bcm47081-tplink-archer-c5-v2.dtb >> /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c5-v2-kernel.bin
/home/harry/openwrt/staging_dir/host/bin/lzma e /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c5-v2-kernel.bin -d16  /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c5-v2-kernel.bin.new
mv /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c5-v2-kernel.bin.new /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c5-v2-kernel.bin
[ -f /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c5-v2-kernel.bin -a -f /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/root.squashfs ]
dd if=/home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/root.squashfs >> /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tplink-archer-c5-v2-squashfs.bin
40114+1 records in
40114+1 records out
20538810 bytes (21 MB, 20 MiB) copied, 0.64125 s, 32.0 MB/s
/home/harry/openwrt/staging_dir/host/bin/trx -o /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tplink-archer-c5-v2-squashfs.bin.trx -m 33554432 -f /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c5-v2-kernel.bin -a 1024
mjn3's trx replacement - v0.81.1
WARNING: maxlen exceeds default maximum!  Beware of overwriting nvram!
/home/harry/openwrt/staging_dir/host/bin/tplink-safeloader -B ARCHER-C5-V2 -k /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tplink-archer-c5-v2-squashfs.bin.trx -r /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tplink-archer-c5-v2-squashfs.bin -j -o /home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tplink-archer-c5-v2-squashfs.bin.new
file-system partition too big (more than 12582912 bytes): Success
Makefile:378: recipe for target '/home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tplink-archer-c5-v2-squashfs.bin' failed
make[5]: *** [/home/harry/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tplink-archer-c5-v2-squashfs.bin] Error 1
make[5]: Leaving directory '/home/harry/openwrt/target/linux/bcm53xx/image'
Makefile:31: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/harry/openwrt/target/linux/bcm53xx'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/harry/openwrt/target/linux'
time: target/linux/install#60.31#68.31#134.49
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/harry/openwrt'
target/Makefile:19: recipe for target '/home/harry/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_install' failed
make[1]: *** [/home/harry/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/harry/openwrt'
/home/harry/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```
我编译的是路由器 Linksys ea6500v2 固件，看上面的提示感觉是编译tplink固件出错了，我要的ea6500固件最后还是生成了，而且刷到路由器里也正常，就是WiFi太差了，掉线还只有802.11bg

avsba001, 2020-02-29:
重新clone下源码，我昨天也报错，删了重新clone源码跑了次编译就正常了

harry3633, 2020-02-29:
> 重新clone下源码，我昨天也报错，删了重新clone源码跑了次编译就正常了

我今天 make clean，git pull，删除 openwrt 后再 clone，都试了，一直出现这个错误，不知道为何固件又都编译出来了，测了下科学上网还挺稳定

closed 2020-03-07 by Lienol