Lienol/openwrt #114: libuClibc++编译错误就是过不去！请Lienol帮看看原因！

zgs360, 2020-02-29:
[image]
@Lienol 大神帮我看看！

zgs360, 2020-02-29:
上在第三张，这是第一张错误开始
[image]
第二张：
[image]
第三张：
[image]

Lienol, 2020-02-29:
把bin,build_dir,staging_dir,tmp目录删掉重新编译试试

zgs360, 2020-02-29:
@Lienol 好的，我重新clone也一样的，也删除过build_dir,staging_dir,tmp,也是一样的，就是bin没删除过，我再试试看！谢谢大神

zgs360, 2020-02-29:
@Lienol 大神还是同样的错误。
删除了：bin,build_dir,staging_dir,tmp
git pull
 ./scripts/feeds update -a
 ./scripts/feeds install -a
 make menuconfig
 make -j1 V=s
错误如下：
[image]

@Lienol 请大神帮我解决！

avsba001, 2020-02-29:
我的解决办法...把openwrt整个文件夹全删除，大概13g，重新跑clone再编译一次就正常了，昨天一样报错

zgs360, 2020-02-29:
@avsba001 我删过几次，clone多次了，还是一样的错误！就是过不了这段！

zgs360, 2020-03-01:
放弃编译，还是用coolsnowwolf大神的ssr plus，

closed 2020-03-08 by Lienol