Lienol/openwrt #121: 如何在外连接路由器ssr使用路由器内的网易云解锁？

WROIATE, 2020-02-29:
2.29日编译固件，如题，根据防火墙规则可以看出网易云解锁是将访问80、443端口的流量全部定向到5200端口进行链接替换，那为什么通过ssr连接路由器却没办法使用呢？

Lienol, 2020-02-29:
需要走路由器自己的OUTPUT链才可以

WROIATE, 2020-03-01:
> 需要走路由器自己的OUTPUT链才可以

链接了ssr服务端后可以走passwall的nat，照理来说也应该可以走网易云啊，为什么不行呢？

Lienol, 2020-03-01:
网易云 没有给路由器（自己）解锁

WROIATE, 2020-03-01:
> 网易云 没有给路由器（自己）解锁

那有没有办法修改防火墙解决？大大能给点建议吗？是需要在mangle里修改吗？

Lienol, 2020-03-01:
输入
iptables -t nat -I OUTPUT 1 -j cloud_music
试试

WROIATE, 2020-03-01:
> 输入
> iptables -t nat -I OUTPUT 1 -j cloud_music
> 试试

不行，用了这条规则以后连正常网页也上不去了，证书全变成网易的了，访问music.163.com显示empty response

WROIATE, 2020-03-01:
> 输入
> iptables -t nat -I OUTPUT 1 -j cloud_music
> 试试

我试着把规则里的1改成2以后发现该走梯子的可以走梯子，然后其余和1一样。网站请求应该是进入了cloud_music这条链以后就出不来了，大大有些什么头绪吗

Lienol, 2020-03-01:
iptables -t nat -I OUTPUT -p tcp -m set --match-set music dst -j cloud_music

WROIATE, 2020-03-01:
> iptables -t nat -I OUTPUT -p tcp -m set --match-set music dst -j cloud_music

这一条我刚才也试了发现可以正常上网易云，歌曲也不灰，但包括正常的所有歌曲都无法播放，所有歌曲都显示加载失败

WROIATE, 2020-03-02:
> iptables -t nat -I OUTPUT -p tcp -m set --match-set music dst -j cloud_music

大大要不你下个版本优化一下？或者给passwall加一个前置代理？

closed 2020-05-16 by Lienol