Lienol/openwrt #127: 自编译的固件版本号和源上 packages 里版本号不一致

CuminLo, 2020-03-02:
### 使用分支
 - dev-19.07

### 自编译固件版本号
`OpenWrt 19.07-SNAPSHOT / LuCI 18-Lienol (git-20.057.51438-539f769)`

### 具体问题
```
opkg install luci
Upgrading luci on root from git-20.057.51438-539f769-1 to git-20.057.55219-13dd17f-1...
Downloading https://openwrt.proxy.ustclug.org/releases/19.07-SNAPSHOT/packages/x86_64/luci/luci_git-20.057.55219-13dd17f-1_all.ipk
Collected errors:
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/releases/19.07-SNAPSHOT/packages/x86_64/luci/luci_git-20.057.55219-13dd17f-1_all.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download luci. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package luci.
```

请求的错误地址： `https://openwrt.proxy.ustclug.org/releases/19.07-SNAPSHOT/packages/x86_64/luci/luci_git-20.057.55219-13dd17f-1_all.ipk` 

网上上正确地址： `https://openwrt.proxy.ustclug.org/releases/19.07-SNAPSHOT/packages/x86_64/luci/luci_git-20.061.59608-041863a-1_all.ipk.ipk` 

我发现是后面的 git 编号不一致。

我编译的版本号是：`20.057.51438-539f769`
下载源上的号码是：`20.061.59608-041863a`

1. 请问如何编译可保持版本号才能一致呢?
2. 我发现把 `https://openwrt.proxy.ustclug.org` 替换成 `https://openwrt.org` 这个地址又是存在的，为啥呢？

接触 `openwrt` 不是很久，期待大佬能解惑。

Lienol, 2020-03-04:
opkg update
为什么还要再装luci？

closed 2020-03-04 by Lienol

CuminLo, 2020-03-06:
@Lienol  opkg update luci 也是一样的提示，请问后面的版本号怎么保持一致？

CuminLo, 2020-03-06:
```
root@OpenWrt:~# opkg list-upgradable
luci-theme-material - git-20.062.36500-0cb6a80-1 - git-20.064.69776-e8c638c-1
luci-mod-rpc - git-20.062.36500-0cb6a80-1 - git-20.064.69776-e8c638c-1
luci-app-statistics - git-20.062.36500-0cb6a80-1 - git-20.064.69776-e8c638c-1
...
```

```
root@OpenWrt:~# opkg upgrade luci-app-aria2
Upgrading luci-app-aria2 on root from 1.0.1-2 to git-20.064.69776-e8c638c-1...
Downloading https://openwrt.proxy.ustclug.org/releases/19.07-SNAPSHOT/packages/x86_64/luci/luci-app-aria2_git-20.064.69776-e8c638c-1_all.ipk
Installing luci-lib-ipkg (git-20.064.69776-e8c638c-1) to root...
Downloading https://openwrt.proxy.ustclug.org/releases/19.07-SNAPSHOT/packages/x86_64/luci/luci-lib-ipkg_git-20.064.69776-e8c638c-1_all.ipk
Collected errors:
 * check_data_file_clashes: Package luci-lib-ipkg wants to install file /usr/lib/lua/luci/model/ipkg.lua
        But that file is already provided by package  * luci-base
```

@Lienol