Lienol/openwrt #130: linksys WRT1900ACv2源码大概有问题

yishan2019, 2020-03-02:
linksys WRT1900ACv2源码大概有问题，刷后重启回到原厂系统，从OpenWrt原版网站下载了这款固件刷后正常进入OpenWrt，因此判断源码有问题

Lienol, 2020-03-02:
用的是哪个分支？

yishan2019, 2020-03-03:
lean-lede

yishan2019, 2020-03-03:
#=================================================
# https://github.com/P3TERX/Actions-OpenWrt
# Description: Build OpenWrt using GitHub Actions
# Lisence: MIT
# Author: P3TERX
# Blog: https://p3terx.com
#=================================================

name: Build OpenWrt

on:
  release:
    types: [published]
  push:
    branches:
      - dev-lean-lede
    paths:
      - '.config'
#  schedule:
#    - cron: 0 8 * * 5
  watch:
    types: [started]

yishan2019, 2020-03-03:
同一天的lean的源码编译出来的能用，但没有passwall

yishan2019, 2020-03-03:
这个分支也试过了不行

env:
  REPO_URL: https://github.com/Lienol/openwrt
  REPO_BRANCH: dev-19.07 
  CONFIG_FILE: .config
  DIY_SH: diy.sh
  FREE_UP_DISK: false
  SSH_ACTIONS: false
  UPLOAD_BIN_DIR: false
  UPLOAD_FIRMWARE: true
  TZ: Asia/Shanghai

closed 2020-03-07 by Lienol