Lienol/openwrt #135: 错误的代码帮忙看看是什么原因

lamu0102, 2020-03-02:
Build dependency: Please install the 'file' package

/home/ldh/openwrt/include/prereq.mk:12: recipe for target 'prereq' failed
Prerequisite check failed. Use FORCE=1 to override.
/home/ldh/openwrt/include/toplevel.mk:176: recipe for target 'staging_dir/host/.prereq-build' failed
make: *** [staging_dir/host/.prereq-build] Error 1
Collecting package info: done
Collecting target info: done
ldh@aml:~/openwrt$

Lienol, 2020-03-02:
Build dependency: Please install the 'file' package

lamu0102, 2020-03-02:
> Build dependency: Please install the 'file' package

ldh@aml:~/openwrt$ /home/ldh/openwrt/include/prereq.mk:12: recipe for target 'prereq' failed
-bash: /home/ldh/openwrt/include/prereq.mk:12:: No such file or directory
ldh@aml:~/openwrt$ Prerequisite check failed. Use FORCE=1 to override.
Prerequisite: command not found
ldh@aml:~/openwrt$ /home/ldh/openwrt/include/toplevel.mk:176: recipe for target 'staging_dir/host/.prereq-build' failed
-bash: /home/ldh/openwrt/include/toplevel.mk:176:: No such file or directory
ldh@aml:~/openwrt$ make: *** [staging_dir/host/.prereq-build] Error 1

lamu0102, 2020-03-02:
ldh@aml:~/openwrt$ /home/ldh/openwrt/include/prereq.mk:12: recipe for target 'prereq' failed
-bash: /home/ldh/openwrt/include/prereq.mk:12:: No such file or directory
ldh@aml:~/openwrt$ Prerequisite check failed. Use FORCE=1 to override.
Prerequisite: command not found
ldh@aml:~/openwrt$ /home/ldh/openwrt/include/toplevel.mk:176: recipe for target 'staging_dir/host/.prereq-build' failed
-bash: /home/ldh/openwrt/include/toplevel.mk:176:: No such file or directory
ldh@aml:~/openwrt$ make: *** [staging_dir/host/.prereq-build] Error 1

lamu0102, 2020-03-02:
> Build dependency: Please install the 'file' package
>
> /home/ldh/openwrt/include/prereq.mk:12: recipe for target 'prereq' failed
> Prerequisite check failed. Use FORCE=1 to override. [...]

lamu0102, 2020-03-02:
sudo apt-get install file  解决了

closed 2020-03-02 by lamu0102