Lienol/openwrt #150: 编译未报错，但是未生成squashfs-sysupgrade.bin文件

cn-404-not-found, 2020-03-05:
反馈bug/问题模板，提建议请删除

## 1.编译未报错，但是未生产squashfs-sysupgrade.bin文件
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：K2P，完全按照教程来做，编译未报错，但是未生成squashfs-sysupgrade.bin文件，只有
openwrt-ramips-mt7621-k2p-initramfs-kernel.bin

### (2) 路由器型号和固件版本

A：K2P，A1版本

### (3) 详细日志

A：无报错

DSNASF, 2020-03-05:
少选择点功能，你这是插件选多了包太大超出了，你kernel的包肯定接近16兆

Lienol, 2020-03-06:
估計樓上正解

cn-404-not-found, 2020-03-06:
> 估計樓上正解

的确是这个原因，但是我重新编译，非常精简那种，就几个插件，编译出来的大小都有15928605，接近16M了，我觉得是不是在别的菜单选了什么东西导致占了空间。

我现在上传了config文件，请给位大佬帮忙看看，谢谢
[config.zip](https://github.com/Lienol/openwrt/files/4297989/config.zip)

DSNASF, 2020-03-06:
你不觉得你编译的主题有点多吗兄弟插件也不少访客wifi有啥用你也编译建议你保留一个默认主题，一个mtwifi   flow  passwall这样就可以编译了16m凑合吧兄弟

wefuxkgfw, 2020-03-06:
@DSNASF 够刚，够勇。牛皮

copyyour, 2020-03-07:
[config.txt](https://github.com/Lienol/openwrt/files/4301584/config.txt)

大佬帮我看看我的也是这样的情况，我已经减少了很多很多app了，主题也只有一个，但是编译完后，还是没有squashfs-sysupgrade.bin ，openwrt-ramips-mt7621-k2p-initramfs-kernel.bin还是有17mb多。。不知道还能删什么了，非常感谢

closed 2020-03-10 by Lienol