Lienol/openwrt #151: 编译K3失败，可以编译成rootfs.tar，就是不能编译成squashfs-sysupgrade.bin，选择就报错

lamu0102, 2020-03-05:
mv /home/ldh/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-netgear-r8000-squashfs.chk.new /home/ldh/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-netgear-r8000-squashfs.chk
file-system partition too big (more than 12582912 bytes): Success
make[5]: *** [/home/ldh/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tmp/openwrt-bcm53xx-tplink-archer-c9-v1-squashfs.bin] Error 1
make[5]: Leaving directory `/home/ldh/openwrt/target/linux/bcm53xx/image'
make[4]: *** [install] Error 2
make[4]: Leaving directory `/home/ldh/openwrt/target/linux/bcm53xx'
make[3]: *** [install] Error 2
make[3]: Leaving directory `/home/ldh/openwrt/target/linux'
time: target/linux/install#159.53#40.45#646.62
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory `/home/ldh/openwrt'
make[1]: *** [/home/ldh/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory `/home/ldh/openwrt'
Build failed - please re-run with -j1 to see the real error message

Lienol, 2020-03-06:
用最新源码试试，不知道为什么K3编译会把所有bcm53xx都给一起编译了

viagram, 2020-03-07:
> 用最新源码试试，不知道为什么K3编译会把所有bcm53xx都给一起编译了

的确是会把bcm53xx的都编译，我都是把除k3以外的都注释掉才行。

closed 2020-03-11 by Lienol