Lienol/openwrt #162: nat游戏模式 PS4 XBOX从3月6日版本开始又不稳定了！！

wzd1016, 2020-03-08:
3月6日说建议大更新，就更新了，结果nat模式在玩游戏时又出现问题，我用白名单模式，把udp和tcp在访问控制（固定ip和mac）里都填写所有,结果xbox的nat模式始终是严格，更新到3月7日偶尔开放，但是之后怎么折腾都严格。我发现能不能去掉那个china-ng啊，那个根本没用，还拖慢速度，我感觉是加了那个导致的，那个真的不稳定啊，套什么dns啊，没用的,smartdns连爱奇艺都会被误以为国外，求求封杀smartdns吧，哎，多说无力。这个问题应该很好解决的，但是为啥总会出现这个问题呢。

9o8i7u, 2020-03-08:
smartdns、adguard、dnscrypt-proxy、dnsforwarder 这类三方DNS不会使用，建议就不要开启，直接使用dnsmasq、pdns类就好。DNS服务器直接 aliyun、114、国外google dns就好。省得因为不会使用，出奇怪的问题。

wzd1016, 2020-03-08:
guan'jian'sguanjians

> smartdns、adguard、dnscrypt-proxy、dnsforwarder 这类三方DNS不会使用，建议就不要开启，直接使用dnsmasq、pdns类就好。DNS服务器直接 aliyun、114、国外google dns就好。省得因为不会使用，出奇怪的问题。

我也没用smartdns啊，关键是游戏机的nat模式不能用啊，我觉得这个是个很简单的问题，之前一端时间也没问题了，最近加回china-ng(这个给smartdns嵌套产生机会的东西）了,nat模式玩游戏又出现问题了，怎么也不开放啊，我怀疑是china-ng在作怪，因为之前版本连国内dns都无法改，那时候没这个问题！

wzd1016, 2020-03-08:
注明，3月3日，就是没那么多乱起八糟，国内国外dns的时候，没这个问题。我刷回3月3日了

jones-76, 2020-03-08:
如果使用smartdns，国内DNS直接默认就行了，不需要非要使用chinaDNS-NG，DNS模式直接本机7913，nat游戏模式我这里TCP,UDP访问控制直接所有没问题，xbox和ps4显示都是全开。

Lienol, 2020-03-08:
没有动过防火墙规则
大概是你装了什么插件冲突了

closed 2020-03-08 by Lienol