Lienol/openwrt #174: 既udp游戏模式后，新的玄学问题（youtube tv同样的节点显示国外）

wzd1016, 2020-03-10:
我一直尝试测试新固件，最新的是3月9日的稳定版和激进版64位，软路由pve虚拟机下的openwrt+ros。昨天说的是xbox 很难udp模式下nat显示开放，您说是玄学或者安了什么插件，在3月6号以后的版本（国内国外dns再次分开后），今天遇到新的问题。同样的机场节点（某一国家vps节点），旧的固件可以看youtube tv, 但是3月9号的固件youtube tv显示不在此国家不让观看,刚开始还赖机场说节点出问题了，可是刷回旧固件（3月3日64位稳定版）就解决了，我不信邪，又刷了2月1日64位稳定版，那时候还有smartdns也分dns国内国外，都没有这样的问题，而且xbox udp也容易转发成功，也就是nat显示开放而不是受限。两件事结合起来我认为新固件有一定的问题，不完全是玄学问题，因为不能让我自己一个人全赶上了

kulanchong, 2020-03-10:
哈，别较真，我提交得bug也是被说玄学，我用了j1900 软路由，n1，newifi3，三个设备新刷机测试出来得bug，提交得时候就写了一个n1，然后大神说是玄学，花了2个小时测试。。。。。。

Lienol, 2020-03-10:
输入命令
iptables -t mangle -L
来看看

Lienol, 2020-03-10:
还有
lsmod | grep TPROXY

Lienol, 2020-03-10:
还有日志

wzd1016, 2020-03-10:
[2月1日64位按照您的smartdns版本.txt](https://github.com/Lienol/openwrt/files/4312393/2.1.64.smartdns.txt)
[3月9日稳定版64位.txt](https://github.com/Lienol/openwrt/files/4312394/3.9.64.txt)

wzd1016, 2020-03-10:
按照您的要求给您写上了，我也不会写程序我也不懂哈哈，2月1日是一切正常版本，3月9日是不正常版本（youtube tv同样节点显示在国外以及xbox 总是显示严格换dns国外模式也不管用）

> 还有日志

ntgeralt, 2020-03-18:
udp转发归转发，对游戏没什么效果。反正我开了udp转发，PS4依旧nat3就是了。这点和ssrplus+一模一样。。

koolshare和hiboy的Padavan udp转发能让ps4从nat3提升到nat2.

另外手动添加节点的按钮能否放在节点之前，减少下鼠标滚轮操作。保存应用也是。

closed 2020-03-22 by Lienol