Lienol/openwrt #183: passwall订阅更新后 默认TCP连接节点会改变

hiwind123, 2020-03-13:
passwall
版本36.21

2020-03-13 10:05:16 订阅后 默认的节点 换掉了 

2020-03-13 10:05:00: 清空并关闭相关程序和缓存完成。
2020-03-13 10:05:00: TCP_1节点：香港HGC-1Gbps(1)，节点：*********************，监听端口：1041
2020-03-13 10:05:00: UDP_1节点：香港HGC-1Gbps(1)，节点：*********************，监听端口：1051
2020-03-13 10:05:00: DNS：pdnsd...
2020-03-13 10:05:00: dnsmasq：生成配置文件。
2020-03-13 10:05:00: 开始加载防火墙规则...
2020-03-13 10:05:01: 访问控制：MAC：******************，使用TCP_1节点，UDP_1节点，大陆白名单
2020-03-13 10:05:01: 访问控制：IP：192.168.8.91，MAC：******************，使用TCP_1节点，UDP_1节点，大陆白名单
2020-03-13 10:05:01: 访问控制：IP：192.168.8.68，MAC：******************，使用TCP_1节点，UDP_1节点，GFW
2020-03-13 10:05:01: 访问控制：MAC：******************，使用TCP_1节点，UDP_1节点，GFW
2020-03-13 10:05:01: 访问控制：IP：192.168.8.148，MAC：******************，使用TCP_1节点，UDP_1节点，GFW
2020-03-13 10:05:01: 访问控制：MAC：DC:71:96:89:AC:97，使用TCP_1节点，UDP_1节点，GFW
2020-03-13 10:05:01: 开始过滤所有IPV4节点到白名单
2020-03-13 10:05:01: 过滤所有IPV4节点完成
2020-03-13 10:05:01: 默认代理模式：不代理
2020-03-13 10:05:01: 防火墙规则加载完成！
2020-03-13 10:05:01: 配置定时任务：自动更新规则。
2020-03-13 10:05:01: 配置定时任务：自动更新节点订阅。
2020-03-13 10:05:01: 运行完成！

2020-03-13 10:05:16: 开始订阅...
2020-03-13 10:05:16: 正在订阅: https://*********************ffline=1
2020-03-13 10:05:23: 成功解析节点数量: 61
2020-03-13 10:05:23: 正在订阅: https://*********************=1
2020-03-13 10:05:25: 成功解析节点数量: 61
2020-03-13 10:05:25: 选择【TCP节点1】第一匹配节点：香港HGC-1Gbps(1)
2020-03-13 10:05:25: 订阅完毕...
2020-03-13 10:05:25: 清空并关闭相关程序和缓存完成。
2020-03-13 10:05:25: TCP_1节点：韩国SK(vpc.kr)，节点：*********************，监听端口：1041
2020-03-13 10:05:25: UDP_1节点：韩国SK(vpc.kr)，节点：*********************，监听端口：1051
2020-03-13 10:05:26: DNS：pdnsd...
2020-03-13 10:05:26: dnsmasq：生成配置文件。
2020-03-13 10:05:26: 开始加载防火墙规则...
2020-03-13 10:05:27: 访问控制：MAC：******************，使用TCP_1节点，UDP_1节点，大陆白名单
2020-03-13 10:05:28: 访问控制：IP：192.168.8.91，MAC：******************，使用TCP_1节点，UDP_1节点，大陆白名单
2020-03-13 10:05:28: 访问控制：IP：192.168.8.68，MAC：******************，使用TCP_1节点，UDP_1节点，GFW
2020-03-13 10:05:28: 访问控制：MAC：******************8，使用TCP_1节点，UDP_1节点，GFW
2020-03-13 10:05:28: 访问控制：IP：192.168.8.148，MAC：******************，使用TCP_1节点，UDP_1节点，GFW
2020-03-13 10:05:28: 访问控制：MAC：DC:71:96:89:AC:97，使用TCP_1节点，UDP_1节点，GFW
2020-03-13 10:05:28: 开始过滤所有IPV4节点到白名单
2020-03-13 10:05:28: 过滤所有IPV4节点完成
2020-03-13 10:05:28: 默认代理模式：不代理
2020-03-13 10:05:28: 防火墙规则加载完成！
2020-03-13 10:05:28: 配置定时任务：自动更新规则。
2020-03-13 10:05:28: 配置定时任务：自动更新节点订阅。
2020-03-13 10:05:28: 运行完成！

luckyyyyy, 2020-03-15:
之前有个版本会这样 换最新的试试 我这边没出现过

luckyyyyy, 2020-03-15:
快看不懂了 测了 是有问题 修一下

closed 2020-03-15 by Lienol

1q1q1, 2020-03-16:
现在在password中已有sock代理，能否在password中增加https代理、以支持caddy插件中的http.forwardproxy直接科学上网？谢谢！

1q1q1, 2020-03-16:
https://jeanniestudio.top/2020/03/14/caddy+http.forwardproxy%20+filebrowser+bbr-%E4%B8%80%E9%94%AE%E6%90%AD%E5%BB%BA/