Lienol/openwrt #187: passwall大陆白名单模式google商店无法连接，能访问google

WROIATE, 2020-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：passwall在大陆白名单开启chinadns使用dns2socks时会导致手机google play客户端无法登陆，但google仍可正常访问，上网查有人说跟google商店udp转发有关，推测是passwall的udp转发有问题，该问题在lean的固件下无

### (2) 路由器型号和固件版本

A：2020.3.14自编译k2p版本

wefuxkgfw, 2020-03-19:
重启手机试试。再不行就手动关闭openwrt的LAN口和WAN口的ipv6相关选项。
https://passwallopenwrt.github.io/website/ipv6.html

WROIATE, 2020-03-19:
> 重启手机试试。再不行就手动关闭openwrt的LAN口和WAN口的ipv6相关选项。
> https://passwallopenwrt.github.io/website/ipv6.html

都试过了，包括关闭ipv6解析，重启手机以及路由器

fonlan, 2020-03-19:
有没有试过把googleapis.cn加域名黑名单然后手机开关一次飞行模式再打开Google Play?

hikaruchang, 2020-03-23:
推荐使用smartdns这类dns解析服务试试

wstxwd007, 2020-03-24:
因为chinadns把国内dns解析的地址发给设备使用，该dns被污染过，所以用不了，直接设置固定解析就OK了

Lienol, 2020-03-24:
测试了好几天了 均未复现
试试更新最新源码

WROIATE, 2020-03-28:
> 测试了好几天了 均未复现
> 试试更新最新源码

昨天用最新源码编译也不行，仍然是可以上谷歌但无法链接谷歌商店
[image]

[image]

closed 2020-04-02 by Lienol