Lienol/openwrt #188: passwall TCP 开启后3秒关闭 之后一只显示未运行

06ec, 2020-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
passwall TCP 开启后3秒关闭 之后一只显示未运行（最后面有.config）
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
自己编译最新的Openwrt（passwall+smartdns+adbaby），编译成功后刷入固件，发现passwall中的TCP开启3秒之后就一只显示未运行，同时无法爬墙。已经编译了三个固件了都是一样的问题。
### (1) 具体问题

A：
固件中，加入了APP库中的smartdns，第一次测试，开启smartdns并设置（期间开启smartdns后，一只显示smartdns-未运行，重启也不行），同时加入爬墙DNS，passwall里面也进行了相应的DNS设置，但是TCP显示未运行；之后第二次测试：重置，设置拨号，设置passwall 开启passwall，TCP运行3秒后，显示未运行，同时无法爬墙。
### (2) 路由器型号和固件版本
A：
K2P A2 
OpenWrt 19.07-SNAPSHOT / LuCI 18-Lienol (git-20.068.65482-506a2be) 
### (3) 详细日志

A：这个日志是未设置和开启smartdns的日志
[日志.txt](https://github.com/Lienol/openwrt/files/4334011/default.txt)
[config.txt](https://github.com/Lienol/openwrt/files/4334012/config.txt)

Lienol, 2020-03-15:
节点类型（SS/SSR/V2RAY/TROJAN/BROOK）是什么？
另外附上passwall日志

06ec, 2020-03-15:
v2ray

2020-03-15 18:58:50: 日志文件过长，清空处理！
2020-03-15 18:58:52: 清空并关闭相关程序和缓存完成。
2020-03-15 18:58:53: TCP_1节点：wc，节点：wc.06ec.xyz:443，监听端口：1041
2020-03-15 18:58:53: DNS：pdnsd...
2020-03-15 18:58:53: dnsmasq：生成配置文件。
2020-03-15 18:58:53: 开始加载防火墙规则...
2020-03-15 18:58:55: 开始过滤所有IPV4节点到白名单
2020-03-15 18:58:55: 过滤所有IPV4节点完成
2020-03-15 18:58:55: 默认代理模式：大陆白名单
2020-03-15 18:58:55: 防火墙规则加载完成！
2020-03-15 18:58:55: 运行完成！

Lienol, 2020-03-15:
看了你的config，应该是编译时没有加入FPU，尝试重新编译，编译时把FPU给勾上重新编译
MIPS_FPU_EMULATOR

06ec, 2020-03-15:
FPU  根据你昨晚在电报群里面发的路径   就没找到  没有这个呀

06ec, 2020-03-15:
还有  smartdns  开启运行后  一直显示未运行  是什么原因呢？

[image]
[image]
[image]

Lienol, 2020-03-15:
DNS话题一律不回复

closed 2020-03-15 by Lienol