Lienol/openwrt #194: 大佬可否帮助看一看下面这个帖子里的问题？万分感谢

sheiy, 2020-03-19:
https://bbs.ikuai8.com/thread-109226-1-1.html
拓扑图如下
[image]

sheiy, 2020-03-19:
[image]
[image]
[image]
[image]
[image]
[image]

liuqi0270, 2020-03-22:
你的问题没有看明白，有什么问题吗？是说你不能富强吗？另外接口编辑中的配置没有看到。

closed 2020-03-22 by Lienol

sheiy, 2020-03-22:
@Lienol 哥们能加个QQ吗？我的QQ35429674，下面这个帖子可能说得清楚一下，麻烦看一下呢，谢谢了，万分感谢，搞了很久了没能解决
https://forums.unraid.net/topic/89952-docker-containers-cant-access-from-the-intelnet/

sheiy, 2020-03-22:
@Lienol 如果我把unraid的网关设置为ikuai就没有问题  unraid里的docker容器都正常上网外网也能正常能访问      网关设置为openwrt容器就不能在外网访问了内网可以