Lienol/openwrt #213: dev-master分支 https-dns-proxy CPU占用率高

lx521603, 2020-03-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
dev-master分支 https-dns-proxy CPU占用率高
Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
dev-master分支 https-dns-proxy CPU占用率高，重启无效，故障依旧，卸载后正常。
### (1) 具体问题

A：dev-master分支 https-dns-proxy CPU占用率高, 4个核心其中1个核心占用率一直100%

### (2) 路由器型号和固件版本

A：软路由，https://github.com/Lienol/openwrt-actions ，云编译X64版本，默认配置，无修改。

### (3) 详细日志

A：

Lienol, 2020-03-26:
不开启也占用吗？

1120296737, 2020-03-26:
这个功能好像是默认开启的，不知道这个功能的功能时，发现占用3865U 50%左右的占用率，建议不默认打开，让有需求的人手动打开

lx521603, 2020-03-26:
> 不开启也占用吗？

Luci里，我根本没勾选，就是没勾选 Luci-https-dns-proxy,

但是在network里面我找到了https-dns-proxy，它默认是勾选的。 并且默认会运行。

我在软件包里卸载的就是这个https-dns-proxy，我早上编译了2个固件都是1个核心100%，其它3个核心正常。

CPU是：Intel(R) Xeon(R) CPU E3-1235L v5 @ 2.00GHz : 4 Core 4 Thread

hanqi7012, 2020-03-27:
同……分配四个核有一个占用到100%

Lienol, 2020-03-27:
直接去掉算了

closed 2020-03-27 by Lienol