Lienol/openwrt #215: 编译报错But that file is already provided by package  * zerotier

WoChen5770, 2020-03-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：选上luci-app-zerotier时报错，发现在network-vpn 有默认选中zerotier，不知道是不是这个问题

### (2) 路由器型号和固件版本

A：X86 64 dev-19.07

### (3) 详细日志

A：Collected errors:
 * check_data_file_clashes: Package luci-app-zerotier wants to install file /home/ichenqiang/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/zerotier
	But that file is already provided by package  * zerotier
 * opkg_install_cmd: Cannot install package luci-app-zerotier.
 * check_data_file_clashes: Package luci-app-zerotier wants to install file /home/ichenqiang/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/zerotier
	But that file is already provided by package  * zerotier
 * opkg_install_cmd: Cannot install package luci-i18n-zerotier-zh-cn.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/ichenqiang/openwrt'
package/Makefile:108: recipe for target '/home/ichenqiang/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/ichenqiang/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/ichenqiang/openwrt'
Build failed - please re-run with -j1 to see the real error message
/home/ichenqiang/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1

Lienol, 2020-03-27:
Build failed - please re-run with -j1 to see the real error message

WoChen5770, 2020-03-27:
这就是j1的报错。就是zerotier 已经生成/etc/init.d/zerotier了，luci-app-zerotier 再去生成就报错了。
我的解决办法是 不勾选luci-app-zerotier，ssh上去把/etc/init.d/zerotier删了 手动安装luci-app-zerotier。

> Build failed - please re-run with -j1 to see the real error message

closed 2020-04-02 by Lienol