Lienol/openwrt #219: make menuconfig出错

ppdragon16, 2020-03-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：首次下载代码make menuconfig

### (2) 路由器型号和固件版本

A：N/A

### (3) 详细日志

A：scripts/feeds update -a有错误：
> Build dependency: Please build with umask 022 - other values produce broken packages
> 
> Prerequisite check failed. Use FORCE=1 to override.
> make: *** [/usr/local/google/home/shuchen/openwrt/include/toplevel.mk:177: staging_dir/host/.prereq-build] Error 1

make menuconfig的错误如下：

> Checking 'working-make'... ok.
> Checking 'case-sensitive-fs'... ok.
> Checking 'proper-umask'... failed.
> Checking 'gcc'... ok.
> Checking 'working-gcc'... ok.
> Checking 'g++'... ok.
> Checking 'working-g++'... ok.
> Checking 'ncurses'... ok.
> Checking 'perl-thread-queue'... ok.
> Checking 'tar'... ok.
> Checking 'find'... ok.
> Checking 'bash'... ok.
> Checking 'patch'... ok.
> Checking 'diff'... ok.
> Checking 'cp'... ok.
> Checking 'seq'... ok.
> Checking 'awk'... ok.
> Checking 'grep'... ok.
> Checking 'getopt'... ok.
> Checking 'stat'... ok.
> Checking 'unzip'... ok.
> Checking 'bzip2'... ok.
> Checking 'wget'... ok.
> Checking 'perl'... ok.
> Checking 'python3-cleanup'... ok.
> Checking 'python'... ok.
> Checking 'git'... ok.
> Checking 'file'... ok.
> Checking 'ldconfig-stub'... ok.
> 
> Build dependency: Please build with umask 022 - other values produce broken packages
> 
> Prerequisite check failed. Use FORCE=1 to override.
> make[1]: *** [/usr/local/google/home/shuchen/openwrt/include/toplevel.mk:177: staging_dir/host/.prereq-build] Error 1
> make: *** [/usr/local/google/home/shuchen/openwrt/include/toplevel.mk:83: prepare-tmpinfo] Error 2
> 
> #### failed to build some targets (1 seconds) ####

ppdragon16, 2020-03-29:
操作系统：Debian rodete，Kernel：5.2.17-1-rodete3-amd64

ppdragon16, 2020-03-29:
我稍后在Ubuntu 18.04上试试。

Lienol, 2020-04-02:
Prerequisite check failed. Use FORCE=1 to override

closed 2020-04-02 by Lienol