Lienol/openwrt #223: lienol大能否请教一个问题？

norikohxb, 2020-03-29:
请教lienol大大
https://github.com/Lienol/openwrt/tree/dev-lean-lede
和
https://github.com/Lienol/lean-lede
这两个有什么区别？
能否告知？
我怎么觉得这两个是一样的东西啊？
如果有空请告诉我区别谢谢了

lin1123, 2020-03-29:
一个是大雕的源码，一个是经过L大修改的源码分支

norikohxb, 2020-03-29:
你仔细进去看。里面的git clone命令都是大雕的源代码。
找不到passwall了。是不是lienol搞错网页了。

norikohxb, 2020-03-29:
> 一个是大雕的源码，一个是经过L大修改的源码分支

有没有空，你能否点击我给的2个连接进去看一下git clone后的命令，然后再去lean大的编译网页看看命令

这3个命令是否都一样？如果都一样，如何一个是一个是，另外一个是？

lin1123, 2020-03-29:
passwell 有一个单独的分支，feeds里面，update了才有。

norikohxb, 2020-03-29:
使用 git clone https://github.com/coolsnowwolf/lede 命令下载好源代码，然后 cd lede 进入目录

这两个网页现在的命令都是这条，
所有的clone都直接来自lede，
如何会凭空生出 passwall

也就是网页标题对了，内容错了，我不是让你去看内容吗？
git clone -b dev-lean-lede https://github.com/Lienol/openwrt lean-lede
应该是这个内容。才对。

lin1123, 2020-03-29:
使用 git clone https://github.com/coolsnowwolf/lede 命令
使用了这条命令就clone 的大雕的源代码，没有passwell的。链接指向的大雕仓库
https://github.com/Lienol/openwrt/tree/dev-lean-lede 这个仓库是根据大雕源码修改的，编译就带passwell

norikohxb, 2020-03-29:
> 使用 git clone https://github.com/coolsnowwolf/lede 命令
> 使用了这条命令就clone 的大雕的源代码，没有passwell的。链接指向的大雕仓库
> https://github.com/Lienol/openwrt/tree/dev-lean-lede 这个仓库是根据大雕源码修改的，编译就带passwell

这个网页下面的命令（文本里面的命令是错的，网址是对的）
你能否看一下这个网页打开的编译命令的第三句

使用 git clone https://github.com/coolsnowwolf/lede 命令

完全和https://github.com/Lienol/openwrt/tree/dev-lean-lede  没有半毛线关系

norikohxb, 2020-03-29:
lienol估计碰巧修改了 https://github.com/Lienol/openwrt/tree/dev-lean-lede 的网页介绍了，里面的内容是错的。
[image]

lin1123, 2020-03-29:
> lienol估计碰巧修改了 https://github.com/Lienol/openwrt/tree/dev-lean-lede 的网页介绍了，里面的内容是错的。
> [image]

命令错了，你要用L大修改的源码复制页面链接就好了
git clone https://github.com/Lienol/openwrt/tree/dev-lean-lede
这样就可以了，估计是忘记改命令了

norikohxb, 2020-03-29:
> > lienol估计碰巧修改了 https://github.com/Lienol/openwrt/tree/dev-lean-lede 的网页介绍了，里面的内容是错的。
> > [image]
> 
> 命令错了，你要用L大修改的源码复制页面链接就好了
> git clone https://github.com/Lienol/openwrt/tree/dev-lean-lede
> 这样就可以了，估计是忘记改命令了

麻烦你看一下看我自己整的这个命令对不对，应该和你给我的那个等效吧？

norikohxb, 2020-03-29:
> 使用 git clone https://github.com/coolsnowwolf/lede 命令
> 使用了这条命令就clone 的大雕的源代码，没有passwell的。链接指向的大雕仓库
> https://github.com/Lienol/openwrt/tree/dev-lean-lede 这个仓库是根据大雕源码修改的，编译就带passwell

我自己是用这个命令的
git clone -b dev-lean-lede https://github.com/Lienol/openwrt lean-lede
麻烦你帮我看看和你给的命令是否等效

closed 2020-03-29 by Lienol

norikohxb, 2020-03-30:
https://github.com/Lienol/lean-lede
这个网页的内容git clone命令也有错误吧，lienol有空的时候修改一下。
这个就不是lean-lede，这个是纯粹的lean的lede，这个命令也换一下吧