Lienol/openwrt #228: UDP 有问题， 开启了链接不上。

liliass, 2020-03-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [* ] 没有类似的issue

## 2. 详细叙述
UDP 有问题， 开启了链接不上。
UDP默认代理模式 - 全局模式
OpenWrt R20.2.15 GDQ IPv4 v3.1[2020] Compiled by eSir / LuCI Master (git-20.058.13156-2faf063)
PassWall 版本 【 3.6-32】

### (1) 具体问题

A：UDP 有问题， 开启了链接不上。
游戏测试：PUBG PC版。
UDP默认代理模式 - 全局模式

UDP默认代理模式 - 游戏模式白名单 - 可以连接，可是不走代理。

### (2) 路由器型号和固件版本

A：软路由，
OpenWrt R20.2.15 GDQ IPv4 v3.1[2020] Compiled by eSir / LuCI Master (git-20.058.13156-2faf063)
PassWall 版本 【 3.6-32】

### (3) 详细日志

`A：`
```
Mon Mar 30 03:46:08 2020 daemon.warn dnsmasq[24739]: ignoring nameserver 192.168.31.1 - local interface
Mon Mar 30 03:46:08 2020 daemon.info dnsmasq[24739]: using 5585 more nameservers
Mon Mar 30 03:46:08 2020 daemon.info dnsmasq[24739]: read /etc/hosts - 4 addresses
Mon Mar 30 03:46:08 2020 daemon.info dnsmasq[24739]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Mon Mar 30 03:46:08 2020 daemon.info dnsmasq-dhcp[24739]: read /etc/ethers - 0 addresses
Mon Mar 30 03:46:08 2020 user.notice nft-qos-monitor: ACTION=update, MACADDR=34:ce:00:81:45:fe, IPADDR=192.168.31.212, HOSTNAME=lumi-gateway-v3_miio56304477
Mon Mar 30 03:46:08 2020 user.notice nft-qos-dynamic: ACTION=update, MACADDR=34:ce:00:81:45:fe, IPADDR=192.168.31.212, HOSTNAME=lumi-gateway-v3_miio56304477
Mon Mar 30 03:46:08 2020 user.notice nft-qos-monitor: ACTION=update, MACADDR=34:ce:00:81:45:fe, IPADDR=192.168.31.212, HOSTNAME=lumi-gateway-v3_miio56304477
Mon Mar 30 03:46:08 2020 daemon.warn dnsmasq[24739]: Ignoring query from non-local network
Mon Mar 30 03:46:09 2020 daemon.err dnscrypt-proxy[4661]: dnscrypt-proxy Unable to retrieve server certificates
Mon Mar 30 03:46:09 2020 user.notice nft-qos-dynamic: ACTION=update, MACADDR=34:ce:00:81:45:fe, IPADDR=192.168.31.212, HOSTNAME=lumi-gateway-v3_miio56304477
Mon Mar 30 03:46:12 2020 daemon.err uhttpd[3868]: grep: write error: Broken pipe
Mon Mar 30 03:46:27 2020 daemon.info dnscrypt-proxy[4661]: dnscrypt-proxy Refetching server certificates
Mon Mar 30 03:46:42 2020 daemon.err dnscrypt-proxy[4661]: dnscrypt-proxy Unable to retrieve server certificates
Mon Mar 30 03:47:03 2020 daemon.info dnscrypt-proxy[4661]: dnscrypt-proxy Refetching server certificates
Mon Mar 30 03:47:03 2020 daemon.err uhttpd[3868]: --2020-03-30 03:47:03--  https://api.github.com/repos/lienol/openwrt-package/releases/latest
Mon Mar 30 03:47:04 2020 daemon.err uhttpd[3868]: Resolving api.github.com... 140.82.112.6
Mon Mar 30 03:47:04 2020 daemon.err uhttpd[3868]: Connecting to api.github.com|140.82.112.6|:443... connected.
Mon Mar 30 03:47:04 2020 daemon.err uhttpd[3868]: HTTP request sent, awaiting response... 200 OK
Mon Mar 30 03:47:04 2020 daemon.err uhttpd[3868]: Length: 3540 (3.5K) [application/json]
Mon Mar 30 03:47:04 2020 daemon.err uhttpd[3868]: Saving to: 'STDOUT'
Mon Mar 30 03:47:04 2020 daemon.err uhttpd[3868]:
Mon Mar 30 03:47:04 2020 daemon.err uhttpd[3868]:      0K ...                                                   100% 24.1M=0s
Mon Mar 30 03:47:04 2020 daemon.err uhttpd[3868]:
Mon Mar 30 03:47:04 2020 daemon.err uhttpd[3868]: 2020-03-30 03:47:04 (24.1 MB/s) - written to stdout [3540/3540]
Mon Mar 30 03:47:04 2020 daemon.err uhttpd[3868]:
Mon Mar 30 03:47:14 2020 daemon.err uhttpd[3868]: grep: write error: Broken pipe
Mon Mar 30 03:47:18 2020 daemon.err dnscrypt-proxy[4661]: dnscrypt-proxy Unable to retrieve server certificates
Mon Mar 30 03:47:19 2020 daemon.err uhttpd[3868]: grep: write error: Broken pipe
Mon Mar 30 03:47:24 2020 daemon.err uhttpd[3868]: grep: write error: Broken pipe
Mon Mar 30 03:47:29 2020 daemon.err uhttpd[3868]: grep: write error: Broken pipe
Mon Mar 30 03:47:42 2020 daemon.info dnscrypt-proxy[4661]: dnscrypt-proxy Refetching server certificates
Mon Mar 30 03:47:49 2020 daemon.err uhttpd[3868]: grep: write error: Broken pipe
Mon Mar 30 03:47:57 2020 daemon.err dnscrypt-proxy[4661]: dnscrypt-proxy Unable to retrieve server certificates
Mon Mar 30 03:48:04 2020 daemon.err uhttpd[3868]: grep: write error: Broken pipe
Mon Mar 30 03:48:24 2020 daemon.info dnscrypt-proxy[4661]: dnscrypt-proxy Refetching server certificates
Mon Mar 30 03:48:24 2020 daemon.err uhttpd[3868]: grep: write error: Broken pipe
Mon Mar 30 03:48:29 2020 daemon.err uhttpd[3868]: grep: write error: Broken pipe
Mon Mar 30 03:48:39 2020 daemon.err dnscrypt-proxy[4661]: dnscrypt-proxy Unable to retrieve server certificates
[... 166 lines cut ...]

Chain mwan3_ifaces_in (1 references)
target     prot opt source               destination
mwan3_iface_in_wan  all  --  anywhere             anywhere             mark match 0x0/0x3f00

Chain mwan3_ifaces_out (1 references)
target     prot opt source               destination
mwan3_iface_out_wan  all  --  anywhere             anywhere             mark match 0x0/0x3f00

Chain mwan3_policy_balanced (2 references)
target     prot opt source               destination
MARK       all  --  anywhere             anywhere             mark match 0x0/0x3f00 /* wan 3 3 */ MARK xset 0x100/0x3f00
Chain mwan3_rule_https (1 references)
target     prot opt source               destination
MARK       all  --  anywhere             anywhere             mark match 0x0/0x3f00 MARK xset 0x100/0x3f00
MARK       all  --  anywhere             anywhere             mark match 0x100/0x3f00 ! match-set mwan3_sticky_https src,src MARK and 0xffffc0ff
mwan3_policy_balanced  all  --  anywhere             anywhere             mark match 0x0/0x3f00
SET        all  --  anywhere             anywhere             mark match ! 0xfc00/0xfc00 del-set mwan3_sticky_https src,src
SET        all  --  anywhere             anywhere             mark match ! 0xfc00/0xfc00 add-set mwan3_sticky_https src,src

Chain mwan3_rules (1 references)
target     prot opt source               destination
mwan3_rule_https  tcp  --  anywhere             anywhere             multiport sports 0:65535 multiport dports https mark match 0x0/0x3f00 /* https */
mwan3_policy_balanced  all  --  anywhere             anywhere             mark match 0x0/0x3f00 /* default_rule */
```

Lienol, 2020-03-29:
规则是没问题的，是否有多拨环境，尝试把mwan3负载均衡规则删除，有可能和这个冲突

liliass, 2020-03-30:
> 规则是没问题的，是否有多拨环境，尝试把mwan3负载均衡规则删除，有可能和这个冲突

都没开启。 而且重置了后，测试的。

Lienol, 2020-03-30:
我知道，你看看刷完后是不是自带默认mwan3规则，如果是就把mwan3规则全删掉再试

liliass, 2020-03-31:
mwan3 关了也是不行。 UDP还是不起作用

closed 2020-04-02 by Lienol

liliass, 2020-04-03:
为什么关了？还是有问题呢