Lienol/openwrt #230: 编译x86-64固件报错。

jlzsss, 2020-03-30:
使用的是dev-19.07分支，编译npc客户端报错。
自从更新npc客户端后，编译始终报错。

czxinshui, 2020-03-30:
bash: -c: line 0: `rm -f /home/water/openwrt/build_dir/target-x86_64_musl/npc-')/.dep_files /home/water/openwrt/build_dir/target-x86_64_musl/npc-')/.dep_files.1'
make[3]: *** [/home/water/openwrt/build_dir/target-x86_64_musl/npc-')/.prepared_9ff6408ff25fb674edda16df6ab8697d_6664517399ebbbc92a37c5bb081b5c53_check] Error 1
make[3]: Leaving directory `/home/water/openwrt/package/lean/npc'
time: package/lean/npc/compile#0.03#0.00#0.11
make[2]: *** [package/lean/npc/compile] Error 2
make[2]: Leaving directory `/home/water/openwrt'
make[1]: *** [/home/water/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/water/openwrt'
make: *** [world] 错误 2

npc有问题，一直编译不成功

fengmz, 2020-03-30:
try updating package/lean/npc/Makefile:
PKG_VERSION:=$(shell curl --silent "https://api.github.com/repos/ehang-io/nps/releases/latest" | grep -Po '"tag_name": "\K.*?(?=")')

Lienol, 2020-03-31:
update the latest source try again

jlzsss, 2020-03-31:
npc编译成功。

closed 2020-04-01 by Lienol