Lienol/openwrt #231: dev-master x86-64编译不成功

yyong201, 2020-03-31:
求大神帮忙分析分析 **dev-master** 分支编译出错，这个libtorrent注释了也没用。

/home/runner/work/dev-master/dev-master/openwrt/staging_dir/target-x86_64_musl/usr/include/boost/asio/impl/write.hpp:425:54: error: no type named 'executor_type' in 'struct libtorrent::socket_type'
2020-03-30T12:33:40.9124600Z      typedef typename AsyncWriteStream::executor_type executor_type;
2020-03-30T12:33:40.9126346Z                                                       ^~~~~~~~~~~~~
2020-03-30T12:33:42.3793188Z Makefile:937: recipe for target 'http_connection.lo' failed
2020-03-30T12:33:42.3793658Z make[5]: Leaving directory '/home/runner/work/dev-master/dev-master/openwrt/build_dir/target-x86_64_musl/rblibtorrent-1.1.13/src'
2020-03-30T12:33:42.3802494Z make[5]: *** [http_connection.lo] Error 1
2020-03-30T12:33:42.3807818Z make[4]: *** [all-recursive] Error 1
2020-03-30T12:33:42.3811309Z make[3]: *** [/home/runner/work/dev-master/dev-master/openwrt/build_dir/target-x86_64_musl/rblibtorrent-1.1.13/.built] Error 2
2020-03-30T12:33:42.3811927Z Makefile:623: recipe for target 'all-recursive' failed
2020-03-30T12:33:42.3812854Z make[4]: Leaving directory '/home/runner/work/dev-master/dev-master/openwrt/build_dir/target-x86_64_musl/rblibtorrent-1.1.13'
2020-03-30T12:33:42.3813418Z Makefile:74: recipe for target '/home/runner/work/dev-master/dev-master/openwrt/build_dir/target-x86_64_musl/rblibtorrent-1.1.13/.built' failed
2020-03-30T12:33:42.3813917Z make[3]: Leaving directory '/home/runner/work/dev-master/dev-master/openwrt/package/lean/rblibtorrent'
2020-03-30T12:33:42.3814223Z time: package/lean/rblibtorrent/compile#72.60#7.86#79.86
2020-03-30T12:33:42.3817605Z package/Makefile:111: recipe for target 'package/lean/rblibtorrent/compile' failed
2020-03-30T12:33:42.3839840Z make[2]: *** [package/lean/rblibtorrent/compile] Error 2
2020-03-30T12:33:42.3840403Z make[2]: Leaving directory '/home/runner/work/dev-master/dev-master/openwrt'
2020-03-30T12:33:42.3841094Z make[1]: *** [/home/runner/work/dev-master/dev-master/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2020-03-30T12:33:42.3843260Z package/Makefile:107: recipe for target '/home/runner/work/dev-master/dev-master/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
2020-03-30T12:33:42.3849397Z make[1]: Leaving directory '/home/runner/work/dev-master/dev-master/openwrt'

lx521603, 2020-04-03:
make menuconfig
取消rblibtorrent看看

Lienol, 2020-04-05:
用19.07吧

closed 2020-04-06 by Lienol