Lienol/openwrt #237: passwall最后一版望修复个小bug

jhsb, 2020-04-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
就是首页谷歌检测有问题，一直显示检测失败
实则可以正常使用。
[image]

最后一版望修复下这个小问题。
听说大佬要停更了，这里感谢下大佬的付出

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：首页谷歌检测有问题

### (2) 路由器型号和固件版本

A：OpenWrt R20.3.19 / LuCI Master (git-20.085.19816-df13074)

### (3) 详细日志

A： 
2020-04-01 22:35:38: SOCKS5_1节点：深港[HKT][1.0X]，节点：x，监听端口：1081
2020-04-01 22:35:38: TCP_1节点：深港[HKT][1.0X]，节点：x，监听端口：1041
2020-04-01 22:35:38: UDP_1节点：深港[HKT][1.0X]，节点：x，监听端口：1051
2020-04-01 22:35:38: DNS：ChinaDNS-NG + pdnsd(8.8.4.4  8.8.8.8)，国内DNS：223.5.5.5,114.114.114.114
2020-04-01 22:35:38: dnsmasq：生成配置文件。
2020-04-01 22:35:38: 开始加载防火墙规则...
2020-04-01 22:35:39: TCP默认代理模式：GFW
2020-04-01 22:35:39: UDP默认代理模式：大陆白名单
2020-04-01 22:35:39: 开始过滤所有IPV4节点到白名单
2020-04-01 22:35:39: 过滤所有IPV4节点完成
2020-04-01 22:35:39: 防火墙规则加载完成！
2020-04-01 22:35:39: 运行完成！

Lienol, 2020-04-02:
wget google.com
能通吗

jhsb, 2020-04-02:
> wget google.com
> 能通吗

正常
[image]

Lienol, 2020-04-02:
你找到/usr/share/passwall/test.sh脚本，在/usr/bin/wget后面加个 -4

closed 2020-04-04 by Lienol