Lienol/openwrt #238: IPSEC VPN 不工作

nicedayever, 2020-04-01:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
固件版本：2020/04/01 根据 Lienol 仓库编译版本。

A：使用了内置的 IPSec VPN 服务器 （ipsec vpn 多用户版本）远程访问路由。
在打开Passwall的情况下，只能访问国外网站如Google，无法访问国内网站 和路由管理界面。
在关闭Passwall 的情况下，一切网站都不能访问。

ps. 已经开启了内核ipv4转发。
### (2) 路由器型号和固件版本

A：x86 64 软路由

### (3) 详细日志

A：

hello--world, 2020-04-02:
我在防火墙自定义规则添加了以下内容能使用，你可以试试,`192.168.100.0`改为自己设置的
```
iptables -I FORWARD -s 192.168.100.0/24 -j ACCEPT
iptables -I POSTROUTING 1 -s 192.168.100.0/24 -j MASQUERADE -t nat

#如若还不能使用 再添加下面这4条
iptables -I INPUT -m policy --dir in --pol ipsec --proto esp -j ACCEPT
iptables -I FORWARD -m policy --dir in --pol ipsec --proto esp -j ACCEPT
iptables -I FORWARD -m policy --dir out --pol ipsec --proto esp -j ACCEPT
iptables -I OUTPUT -m policy --dir out --pol ipsec --proto esp -j ACCEPT
```

nicedayever, 2020-04-02:
这四个代码解决问题

`iptables -I INPUT -m policy --dir in --pol ipsec --proto esp -j ACCEPT
iptables -I FORWARD -m policy --dir in --pol ipsec --proto esp -j ACCEPT
iptables -I FORWARD -m policy --dir out --pol ipsec --proto esp -j ACCEPT
iptables -I OUTPUT -m policy --dir out --pol ipsec --proto esp -j ACCEPT
`

nicedayever, 2020-04-02:
> 我在防火墙自定义规则添加了以下内容能使用，你可以试试,`192.168.100.0`改为自己设置的
>
> ```
> iptables -I FORWARD -s 192.168.100.0/24 -j ACCEPT
> iptables -I POSTROUTING 1 -s 192.168.100.0/24 -j MASQUERADE -t nat
>
> #如若还不能使用 再添加下面这4条 [...]

谢谢，用后边的代码可以解决问题

Lienol, 2020-04-02:
有没有试过把flow关掉

nicedayever, 2020-04-02:
> 有没有试过把flow关掉

 Flow Offloading 转发加速 一直保持开启状态

Lienol, 2020-04-02:
最新源码已做兼容

closed 2020-04-02 by Lienol

ragouhuo, 2020-08-07:
> 最新源码已做兼容

默认配置，2020.5.5 7baf856 还正常的，之后至今又不行了