Lienol/openwrt #239: 做文件存储时，访问文件的速度过慢

nicedayever, 2020-04-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

软路由的文件下载速度过慢，无论是 WebDAV，SMB，还是 普通HTTP 速度下载，都只有不到1 MBps，以前速度可以达到50MBps。

一种测试方法：
在 `/tmp` 路径下 创建名为 `test`的1G测试文件，使用 `python2 -m SimpleHTTPServer 8080` 创建web服务端口，用浏览器测试下载速度。实际下载速度仅为0.8 MBps左右。

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

版本号：OpenWrt 19.07-SNAPSHOT / LuCI 18-Lienol (git-20.089.84594-ca183e3)  2020/04/01 编译
硬件： J1900 软路由

### (3) 详细日志

A：

nicedayever, 2020-04-03:
问题原因

关闭 bbr 加速后，速度直接满速。
开启 bbr 加速后，速度又变为 <1MBps。

其他选项 无影响。

受到影响的应用：

    SMB
    WebDAV
    几乎所有和本地文件 -> 外网传输的 都会受到影响。

avsba001, 2020-04-11:
我测试了下，没这个问题，BBR有段时间没更新了吧
测试环境：
dev-master分支
黑群辉 web smb 双测试均达上传带宽满速
网络 软路由所在电信  测试网络为移动和联通

closed 2020-05-06 by Lienol

koala0529, 2020-11-07:
我也遇到了这个问题，我是MediaTek MT7621 ver:1 eco:3
装上bbr写速度正常，读取速度很慢就3兆左右
有没有解决办法？