Lienol/openwrt #242: 这版master分支编译了，还是没见温度

yishan2019, 2020-04-03:
这版master分支编译了，还是没见温度

Lienol, 2020-04-03:
什么CPU，传感器选了吗

yishan2019, 2020-04-03:
我用D2550试的，传感器选了，明天再用我的1900试试。不过以前编译一次它们都有温度显示，也不用刻意选默认的就有，这次还刻意选了

Lienol, 2020-04-03:
控制台输入 sensors 看看

yishan2019, 2020-04-03:
控制台
有显示，但路由器界面没有显示

Lienol, 2020-04-03:
输出内容截下图 和 首页也截下图

yishan2019, 2020-04-03:
输入sensors后昨晚睡了没看路由器页面，今早来截图看有温度了！是不是以后刷机后都要输入sensors命令呢？

yishan2019, 2020-04-03:
下面时控制台的：
root@OpenWrtsyb:~# sensors
acpitz-acpi-0
Adapter: ACPI interface
temp1:        +26.8°C  (crit = +100.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Core 0:       +35.0°C  (crit = +100.0°C)
Core 1:       +34.0°C  (crit = +100.0°C)

root@OpenWrtsyb:~#

yishan2019, 2020-04-03:
主页面的：
状态
系统
主机名	OpenWrtsyb
型号	Intel(R) Atom(TM) CPU D2550 @ 1.86GHz : 2 Core 4 Thread
CPU信息	1861.983 MHz +35.0°C
固件版本	OpenWrt SNAPSHOT / LuCI Master-Lienol git-20.093.65242-67bff18
内核版本	5.4.28
本地时间	2020-04-04 07:55:26
运行时间	9h 21m 51s
平均负载	0.00, 0.02, 0.00

Lienol, 2020-04-03:
。。。
估计浏览器有缓存

closed 2020-04-04 by Lienol