Lienol/openwrt #257: 编译失败

Sunnice, 2020-04-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue 
无

## 2. 详细叙述
使用lienol大佬的dev-lean-lede源码编译竞斗云2.0的固件。出现了几个错误导致编译失败，日志已附上。

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：竞斗云2.0 

### (3) 详细日志

A：2020-04-06T09:18:52.3823229Z  make[3] -C feeds/luci/collections/luci-ssl-openssl compile
2020-04-06T09:18:56.8496800Z  make[3] -C package/lean/luci-app-clash compile
2020-04-06T09:18:59.5751542Z  make[3] -C package/lean/luci-app-qbittorrent compile
2020-04-06T09:29:17.0202084Z  make[3] -C package/lean/UnblockNeteaseMusic compile
2020-04-06T09:29:18.0283532Z  make[3] -C package/lean/luci-app-unblockmusic compile
2020-04-06T09:29:18.8805581Z  make[2] package/install
2020-04-06T09:29:38.1306047Z make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
2020-04-06T09:29:38.1307670Z make: *** [world] Error 1
2020-04-06T09:29:38.1308141Z /home/runner/work/actions-openwrt-passwall/actions-openwrt-passwall/openwrt/include/toplevel.mk:225: recipe for target 'world' failed

2020-04-06T09:32:04.3372944Z xshared.c: In function 'parse_wait_time':
2020-04-06T09:32:04.3373967Z xshared.c:413:13: warning: inlining failed in call to 'xs_has_arg': call is unlikely and code size would grow [-Winline]
2020-04-06T09:32:04.3374390Z  inline bool xs_has_arg(int argc, char *argv[])
2020-04-06T09:32:04.3374696Z              ^~~~~~~~~~
2020-04-06T09:32:04.3374987Z xshared.c:327:13: note: called from here
2020-04-06T09:32:04.3375379Z   } else if (xs_has_arg(argc, argv))
2020-04-06T09:32:04.3375643Z              ^~~~~~~~~~~~~~~~~~~~~~
2020-04-06T09:32:04.3418101Z xshared.c: In function 'parse_wait_interval':
2020-04-06T09:32:04.3418965Z xshared.c:413:13: warning: inlining failed in call to 'xs_has_arg': call is unlikely and code size would grow [-Winline]
2020-04-06T09:32:04.3419358Z  inline bool xs_has_arg(int argc, char *argv[])
2020-04-06T09:32:04.3419647Z              ^~~~~~~~~~
2020-04-06T09:32:04.3419932Z xshared.c:343:11: note: called from here
2020-04-06T09:32:04.3420211Z   else if (xs_has_arg(argc, argv))
2020-04-06T09:32:04.3420496Z            ^~~~~~~~~~~~~~~~~~~~~~
2020-04-06T09:32:04.4232876Z mv -f .deps/xtables_legacy_multi-xshared.Tpo .deps/xtables_legacy_multi-xshared.Po
2020-04-06T09:32:04.4267303Z arm-openwrt-linux-muslgnueabi-gcc -DHAVE_CONFIG_H -I. -I..  -D_LARGEFILE_SOURCE=1 -D_LARGE_FILES -D_FILE_OFFSET_BITS=64 -D_REENTRANT 	-

2020-04-06T09:42:36.8667906Z Configuring luci-i18n-nlbwmon-zh-cn.
2020-04-06T09:42:36.8693872Z make[2]: *** [package/install] Error 255
2020-04-06T09:42:36.8694307Z package/Makefile:65: recipe for target 'package/install' failed
2020-04-06T09:42:36.8697823Z make[1]: *** [/home/runner/work/actions-openwrt-passwall/actions-openwrt-passwall/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
2020-04-06T09:42:36.8698898Z make[2]: Leaving directory '/home/runner/work/actions-openwrt-passwall/actions-openwrt-passwall/openwrt'
2020-04-06T09:42:36.8699983Z package/Makefile:108: recipe for target '/home/runner/work/actions-openwrt-passwall/actions-openwrt-passwall/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install' failed
2020-04-06T09:42:36.8714482Z make[1]: Leaving directory '/home/runner/work/actions-openwrt-passwall/actions-openwrt-passwall/openwrt'
2020-04-06T09:42:36.8733113Z make: *** [world] Error 2
2020-04-06T09:42:36.8733695Z /home/runner/work/actions-openwrt-passwall/actions-openwrt-passwall/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
2020-04-06T09:42:36.8872724Z ##[error]Process completed with exit code 2.

imnotpopo, 2020-04-06:
几线程编译的？梯子没问题的话试一下 make -j1 V=s

Sunnice, 2020-04-07:
> 几线程编译的？梯子没问题的话试一下 make -j1 V=s

云编译，配置文件里面make -j1 V=s

closed 2020-04-14 by Lienol