Lienol/openwrt #259: 无法挂载overlay

flyaway318, 2020-04-06:
1907固件空间太小，建议增大容量.挂载硬盘扩容，无法挂载overlay，请改进检查谢谢

lin1123, 2020-04-06:
什么意思，编译的时候固件空间默认160m，可以自己调整

flyaway318, 2020-04-06:
挂载overlay扩容不好用，挂载不上

nicedayever, 2020-04-13:
> 挂载overlay扩容不好用，挂载不上

同样挂载不上，只能靠自己调整 空间

ppdragon16, 2020-04-18:
+1.

[image]

Lienol, 2020-05-06:
[image]
[image]

Lienol, 2020-05-06:
更新源码再试试

chenxudong2020, 2020-05-07:
应该加个配置 每个路由器都不同的ROM 像我这个B70 256NAND Flash 刷机后默认最大才9M 这个不正常啊

ppdragon16, 2020-05-07:
现在更新源码是不是pw就没了啊？

closed 2020-05-10 by Lienol

pzxiaoyi, 2020-06-22:
> 什么意思，编译的时候固件空间默认160m，可以自己调整

请问如何调整大一点呢？32G的硬盘，不用可惜了。

zhouzl7, 2020-07-30:
> 更新源码再试试

挂载不上啊，网上的所有方法都试过了。你这是怎么做到的？有教程吗

wang1zhen, 2020-10-30:
> 
> 
> 什么意思，编译的时候固件空间默认160m，可以自己调整

您好，请问怎样在编译时调整overlay分区的大小呢？我没有找到相关的编译选项。

zsf020, 2020-12-03:
OpenWrt 19.07-SNAPSHOT / LuCI 17.01 Lienol (git-78679eb)overlay挂载不成功，要在磁盘管理那里手动挂载SDA3为/overlay才可以，但重启后又失效，要重复在磁盘管理那里手动挂载SDA3为/overlay才可以。

zsf020, 2020-12-03:
[
[image]