Lienol/openwrt #265: 竞斗云2.0 p2w_r619ac-128m 2.4Gwifi的LED指示灯不亮，可以修复一下吗？lean版的LED正常

doney0318, 2020-04-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
A：2.4Gwifi的LED指示灯不亮，LEAN版的LED正常
LED 名称： r619ac:blue:wlan2g
触发器：phy0tpt

### (2) 路由器型号和固件版本
A：竞斗云2.0 p2w_r619ac-128m / OpenWrt 19.07-SNAPSHOT / LuCI 18-Lienol (git-20.100.45594-77e2cc6)

hyird, 2020-04-12:
以前亮么

doney0318, 2020-04-12:
使用 lienol dev-19.07编译的固件就从来没同时亮过两个WIFI的LED，都是只亮一个，使用 lean 的编译wifi的LED灯都正常

hyird, 2020-04-12:
> 使用 lienol dev-19.07编译的固件就从来没同时亮过两个WIFI的LED，都是只亮一个，使用 lean 的编译wifi的LED灯都正常

我去问了下大佬 修这个问题的patch  400、851、852、853这几个patch都有啊 不应该呀

doney0318, 2020-04-12:
https://github.com/coolsnowwolf/lede/tree/master/target/linux/ipq40xx
我用这里的包替换lienol 上的patch 可以修复吗

hyird, 2020-04-12:
我看了下 该有的都有了 其他问题我就不知道了

doney0318, 2020-04-12:
使用以下
echo "0" > /sys/class/leds/r619ac:blue:wlan5g/brightness
echo "1" > /sys/class/leds/r619ac:blue:wlan5g/brightness
5G灯可以正常开关
echo "1" > /sys/class/leds/r619ac:blue:wlan2g/brightness
echo "0" > /sys/class/leds/r619ac:blue:wlan2g/brightness
2G无任何效果

hyird, 2020-04-13:
> 使用以下
> echo "0" > /sys/class/leds/r619ac:blue:wlan5g/brightness
> echo "1" > /sys/class/leds/r619ac:blue:wlan5g/brightness
> 5G灯可以正常开关
> echo "1" > /sys/class/leds/r619ac:blue:wlan2g/brightness
> echo "0" > /sys/class/leds/r619ac:blue:wlan2g/brightness
> 2G无任何效果

这是肯定的啦

sujianchao, 2020-04-15:
这问题我也遇到了。无论是用电报群上的固件还是自己编辑最新的源码，2.4G的灯还是不亮。这问题已经出现好久了，希望能得到修复。

hyird, 2020-04-17:
这个问题大佬没机子无法排查

doney0318, 2020-04-17:
lean 版的是好的，而且他的源码已打上 patches-4.19 的补丁，这里的还一直是patches-4.14，可以同步一下补丁吗？

hyird, 2020-04-17:
> lean 版的是好的，而且他的源码已打上 patches-4.19 的补丁，这里的还一直是patches-4.14，可以同步一下补丁吗？
你要弄自己弄 files-4.19 patch-4.19 然后makefile里面改成4.19 好像还有个地方 也得改
内核这东西稳定就行了 没有必要最新 4.19比4.14多了什么我相信大部分人都不知道 我也不知道。。
路由这东西稳定第一

PR #293 (2020-04-24): ipq40xx: fixed some bug
1. fix led and switch config after commit 180fc84ed057846d6288ba22ebbff740725915fa
2. fix caldata and sysupgrade for Asus RT-Acrh 17, fix #285
3. add missing config for ipq4019 sd driver, fix #265 and revert #264

@hyird 来测试下？
commit 7eef5d96 (2020-04-24): ipq40xx: fixed some bug (#293)

1. fix led and switch config after commit 180fc84ed057846d6288ba22ebbff740725915fa
2. fix caldata and sysupgrade for Asus RT-Acrh 17, fix #285
3. add missing config for ipq4019 sd driver, fix #265 and revert #264
closed 2020-04-24 by Lienol