Lienol/openwrt #274: 竞斗云2.0固件的交换机创建vlan2后，wan口连接失败（交换机配置中wan口缺失），如何修复？

sujianchao, 2020-04-15:
### (1) 具体问题
竞斗云2.0固件在“交换机”配置中创建多vlan后，wlan就断开连接上不了！经过查找network配置中发现通过luci添加vlan会默认创建vlan2、vlan3、vlan4这样依次创建，但是vlan2默认是隐性的已经分配给wlan上网端口，所以每次通过luci创建vlan都会首先创建vlan2，导致创建后就wlan断开了链接。麻烦修复。
交换机默认配置都只显示	CPU (eth0)	LAN 1	LAN 2	LAN 3	LAN 4，并没有看到wan端口的配置，麻烦也一并修复了吧。
[有问题的交换机截图](https://wx2.sbimg.cn/2020/04/15/switch_err.png)
[正常的截图](https://wx2.sbimg.cn/2020/04/15/switch_ok.png)

### (2) 路由器型号和固件版本
openwrt-ipq40xx-generic-p2w_r619ac-128m-squashfs-nand-sysupgrade.bin

hyird, 2020-04-17:
一般对wan没有操作 你也用不着吧  默认是cpu未标记 他是不生效的  你要lan改wan需要将默认的删除

sujianchao, 2020-04-17:
> 一般对wan没有操作 你也用不着吧 默认是cpu未标记 他是不生效的 你要lan改wan需要将默认的删除

我需要做iptv的单线复用，vlan1默认上网lan，vlan2连光猫上网拨号端口，vlan3连光猫的iptv端口，上面图二我刷其他固件配置是正常的，但是用这仓库编译的固件就是不能配置wan，所以进行不下去。
     虽然我参照正常的network配置文件通过ssh命令行配置已经把这问题解决了，但是如果仅仅用luci的web页面是配置不了wan的，所以还是需要反馈下。

hyird, 2020-04-17:
这个我就不会了 尝试改过 wan到是能显示了 不过默认的接口绑定又有问题

sujianchao, 2020-04-22:
首先非常感谢L大@Lienol 在这次 (https://github.com/Lienol/openwrt/commit/87c552e351f8ef6ddc833fa30a43e28c20cda9d1) 提交后交换机上终于有wan口配置了，但是有个很致命的bug就是默认生成的WAN口物理接口连接的是eth0.2（也即是交换机的vlan2）是上不了网的，只能按旧版的配置，将wan连接到"eth1"才能上网。
我自己尝试修改了“02_network”源码编译过几次，还是一样的问题，然后我再和lede的固件做过对比，发现lede的固件编译后是没有出现“eth1”这个接口的，默认的“eth0.2”是上网wan口，正常。但是这个源码是会同时出现“eth0”、”eth1“，两个接口，并且默认的“eth0.2”作为上网wan口不正常。希望L大有空可以跟进这bug。

hyird, 2020-04-24:
> 首先非常感谢L大@Lienol 在这次 ([87c552e](https://github.com/Lienol/openwrt/commit/87c552e351f8ef6ddc833fa30a43e28c20cda9d1)) 提交后交换机上终于有wan口配置了，但是有个很致命的bug就是默认生成的WAN口物理接口连接的是eth0.2（也即是交换机的vlan2）是上不了网的，只能按旧版的配置，将wan连接到"eth1"才能上网。
> 我自己尝试修改了“02_network”源码编译过几次，还是一样的问题，然后我再和lede的固件做过对比，发现lede的固件编译后是没有出现“eth1”这个接口的，默认的“eth0.2”是上网wan口，正常。但是这个源码是会同时出现“eth0”、”eth1“，两个接口，并且默认的“eth0.2”作为上网wan口不正常。希望L大有空可以跟进这bug。
> [image]
> 
> [image]
> 
> [image]

大佬已经来修复了 坐等合并

sujianchao, 2020-04-24:
> 大佬已经来修复了 坐等合并
 感谢L大 @Lienol 终于修复了，经测试，暂未发现问题。https://github.com/Lienol/openwrt/commit/180fc84ed057846d6288ba22ebbff740725915fa

closed 2020-04-24 by Lienol