Lienol/openwrt #279: li大你好编译失败 ，请指教

qq1316237299, 2020-04-16:
[image]
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-passwall:
 * 	chinadns-ng
 * opkg_install_cmd: Cannot install package luci-app-passwall.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/qq1316237299/lede'
package/Makefile:108: recipe for target '/home/qq1316237299/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/qq1316237299/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/qq1316237299/lede'
/home/qq1316237299/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

- [ ] 

- [ ] 
[image]

hyird, 2020-04-17:
缺了依赖

qq1316237299, 2020-04-17:
> 缺了依赖

嗯嗯 好了 我以为就加入一个passwall的源码包就好了

closed 2020-04-17 by Lienol

hellstars, 2020-06-12:
我同样的问题 
[image]