Lienol/openwrt #292: dev-master分支编译错误！请指教！

shenzt68, 2020-04-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：dev-master分支编译错误

### (2) 路由器型号和固件版本

A：编译N1固件，用armv8

### (3) 详细日志

install -d -m0755 /home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17/ipkg-aarch64_generic/python-base/usr/bin /home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17/ipkg-aarch64_generic/python-base/usr/lib
ln -sf python2.7 /home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17/ipkg-aarch64_generic/python-base/usr/bin/python
ln -sf python2.7 /home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17/ipkg-aarch64_generic/python-base/usr/bin/python2
cp -fpR /home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17/ipkg-install/usr/lib/libpython2.7.so* /home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17/ipkg-aarch64_generic/python-base/usr/lib/
/usr/bin/env bash ../python-package-install.sh "2" "/home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17/ipkg-install" "/home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17/ipkg-aarch64_generic/python-base" "/home/szt/openwrt/staging_dir/hostpkg/bin/python2.7" "" "$V_PyPackage_python_base_filespec" && if [ -d "/home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17/ipkg-aarch64_generic/python-base/usr/bin" ]; then /home/szt/openwrt/staging_dir/host/bin/sed -i -e "1"'!'"b;s,^#"'!'".*python.*,#"'!'"/usr/bin/python2," -i --follow-symlinks /home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17/ipkg-aarch64_generic/python-base/usr/bin/* ; fi
/home/szt/openwrt/staging_dir/host/bin/sed: cannot stat /home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17/ipkg-aarch64_generic/python-base/usr/bin/python2.7: No such file or directory
Makefile:375: recipe for target '/home/szt/openwrt/bin/packages/aarch64_generic/packages/python-base_2.7.17-2_aarch64_generic.ipk' failed
make[3]: *** [/home/szt/openwrt/bin/packages/aarch64_generic/packages/python-base_2.7.17-2_aarch64_generic.ipk] Error 4
make[3]: Leaving directory '/home/szt/openwrt/feeds/packages/lang/python/python'
time: package/feeds/packages/python/compile#0.33#0.17#0.44
package/Makefile:111: recipe for target 'package/feeds/packages/python/compile' failed
make[2]: *** [package/feeds/packages/python/compile] Error 2
make[2]: Leaving directory '/home/szt/openwrt'
package/Makefile:107: recipe for target '/home/szt/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: *** [/home/szt/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/szt/openwrt'
/home/szt/openwrt/include/toplevel.mk:218: recipe for target 'world' failed
make: *** [world] Error 2

weiwuji1, 2020-04-24:
dev-master分支编译x86-64遇到同样问题，求指教

Lienol, 2020-04-25:
python2已经被官方遗弃

hyird, 2020-04-25:
/home/szt/openwrt/build_dir/target-aarch64_generic_musl/Python-2.7.17

这个目录删了 重新跑就好了

weiwuji1, 2020-04-26:
感谢，删除后编译成功

hyird, 2020-04-26:
问题搞定 issue关了啊

weiwuji1, 2020-04-27:
不是我发起的啊，不知道其他人有没有解决

closed 2020-04-29 by Lienol