Lienol/openwrt #301: docker无法运行容器

lexin8, 2020-04-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
已经挂载 /opt 目录
### (1) 具体问题
docker能正常拉取镜像，安装容器就报错无法运行，尝试其他镜像也是一样
A：

### (2) 路由器型号和固件版本
OpenWrt 19.07-SNAPSHOT / LuCI 18-Lienol (git-20.118.52134-65913a0)
内核版本	4.14.176
X86平台
A：

### (3) 详细日志
root@OpenWrt:~# docker run -d --restart=always --name="portainer" -p 9999:9000 -v /var/run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer
39bcb5f2f378c67116fd97c3f4732b99dc71a39100d62075859b27afefff5ba0
docker: Error response from daemon: OCI runtime create failed: container_linux.go:346: starting container process caused "process_linux.go:449: container init caused \"rootfs_linux.go:58: mounting \\\"mqueue\\\" to rootfs \\\"/opt/docker/overlay2/30f495eae1c71c07930d3bdc11bac06df5a97c22d8c9b165d3e70cb269b9a43e/merged\\\" at \\\"/dev/mqueue\\\" caused \\\"no such device\\\"\"": unknown.

A：

lexin8, 2020-05-15:
据说是内核版本不兼容，dev-master代码貌似正常

Lienol, 2020-05-15:
缺少 POSIX_MQUEUE 内核
编译时加入即可
我抽空优化一下

Lienol, 2020-05-15:
最新源码已优化，记得也要更新feeds

closed 2020-05-15 by Lienol