Lienol/openwrt #306: 今天的源码版本编译B70固件出错！

chenxudong2020, 2020-05-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：https://github.com/chenxudong2020/HC5962 自动编译出错

### (2) 路由器型号和固件版本

A：HC5960 或者B70

### (3) 详细日志

A：make  -C "/home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.179" HOSTCFLAGS="-O2 -I/home/runner/work/HC5962/HC5962/openwrt/staging_dir/host/include -I/home/runner/work/HC5962/HC5962/openwrt/staging_dir/hostpkg/include -I/home/runner/work/HC5962/HC5962/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat May  9 14:26:48 2020" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/runner/work/HC5962/HC5962/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.14.179 CONFIG_MT76x02_LIB=m CONFIG_MT76x2_COMMON=m CONFIG_MT76x2E=m CONFIG_MT7603E=m M="/home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72" NOSTDINC_FLAGS="-I/home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72 -I/home/runner/work/HC5962/HC5962/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport/uapi -I/home/runner/work/HC5962/HC5962/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport -I/home/runner/work/HC5962/HC5962/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/uapi -I/home/runner/work/HC5962/HC5962/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211 -include backport/autoconf.h -include backport/backport.h -DCONFIG_MAC80211_MESH" modules
make[4]: Entering directory '/home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.179'
  CC [M]  /home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/mmio.o
  CC [M]  /home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/util.o
  CC [M]  /home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/trace.o
  CC [M]  /home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/dma.o
  CC [M]  /home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/mac80211.o
/home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/mac80211.c: In function 'mt76_phy_init':
/home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/mac80211.c:286:31: error: 'NL80211_EXT_FEATURE_AQL' undeclared (first use in this function); did you mean 'NL80211_EXT_FEATURE_TXQS'?
  wiphy_ext_feature_set(wiphy, NL80211_EXT_FEATURE_AQL);
                               ^~~~~~~~~~~~~~~~~~~~~~~
                               NL80211_EXT_FEATURE_TXQS
/home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/mac80211.c:286:31: note: each undeclared identifier is reported only once for each function it appears in
/home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/mac80211.c: In function 'mt76_airtime_report':
/home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/mac80211.c:763:12: error: implicit declaration of function 'ieee80211_calc_rx_airtime'; did you mean 'ieee80211_check_tim'? [-Werror=implicit-function-declaration]
  airtime = ieee80211_calc_rx_airtime(dev->hw, &info, len);
            ^~~~~~~~~~~~~~~~~~~~~~~~~
            ieee80211_check_tim
cc1: all warnings being treated as errors
make[5]: *** [/home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/mac80211.o] Error 1
scripts/Makefile.build:326: recipe for target '/home/runner/work/HC5962/HC5962/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-04-27-5f3ccc72/mac80211.o' failed

chenxudong2020, 2020-05-10:
应该是驱动更新了 但是方法没更新 导致的

Lienol, 2020-05-10:
先回滚吧

closed 2020-05-10 by Lienol