Lienol/openwrt #307: 防火墙流量规则调整排序保存报错

roycho93, 2020-05-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：网络-防火墙-流量规则中的规则在向上或者向下排列好后点击保存或者保存并应用就会报错，代码如下

Failed to execute arcombine dispatcher target for entry '/admin/network/firewall/rules'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:1234: bad argument #2 to 'reorder' (string expected, got table)
stack traceback:
	[C]: in function 'reorder'
	/usr/lib/lua/luci/cbi.lua:1234: in function 'parse'
	/usr/lib/lua/luci/model/cbi/firewall/rules.lua:33: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:248: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:387: in function 'parse'
	/usr/lib/lua/luci/dispatcher.lua:860: in function 'target'
	/usr/lib/lua/luci/dispatcher.lua:963: in function </usr/lib/lua/luci/dispatcher.lua:959>

烦请看看是什么原因导致

Lienol, 2020-05-10:
更新feeds源码再试试

roycho93, 2020-05-10:
> 更新feeds源码再试试

感谢大神回复，因为用的别人编译的固件，我麻烦他试试哈。试了很多固件lienol大神的源码的固件最好用，再次致谢。
但是还发现一个不知道是不是问题的小问题，找了很多地方找不到解决方法，所以可能的话lienol大神能解答将不胜感激。
因为需要用的softether中的openvpn，正好我下载别人编译lienol大神源码编译的固件中有softether，设置好后在管理软件中生成ovpn文件时，发现key跟cer那边是空的，ca是能生成，win，ios，安卓跳过没有证书还是啥的提示也姑且可以用，但是mac上就不能用了，连接报错，大概意思是缺少证书之类，于是又用家里别的路由里带的softether测试，生成的文件里是有key跟cer的，唯一发现的就是家里的路由里softether是4.23的老版本，新的是5.01的版本，不知道是bug还是设置问题还是新版取消了这个？
[image]
[image]
[image]

Lienol, 2020-05-15:
软件问题吧

closed 2020-05-15 by Lienol